armel/uv-k5-firmware-custom #100: Bandwidth selection N / W -  5 / 6.25 / 8.33 / 12.5 / 25 KHz

xes, 2024-05-12:
Would it be possible to add more bandwidth values and show the exact number?
For example, others fw allow to choose between 5 / 6.25 / 8.33 / 12.5 / 25 KHz

Where 25 and 12.5 are the usual bandwidths in use for HAM, 
8.33 Khz is used in VHF airband
6.25 KHz is needed to receive/speak with PMR radios (that's why PMR audio is often so low) and receive some marine channels
5 KHz - cannot find specific examples, but some marine weather channels that are very noisy with 6.25, can be received very well with a 5KHz bandwidth.

https://github.com/kamilsss655/uv-k5-firmware-custom/blob/9aa505e9d1e431d82f003a39605594d9a069eb20/driver/bk4819.c#L557

armel, 2024-05-12:
Hi,

Ok, thank you. I'll take a look on it. 

Armel F4HWN.

londresarthur, 2024-07-27:
This is the icing on the cake that is missing for this firmware.

egepmr, 2024-08-28:
Hello,

Thanks a lot for your great firmware. It's my favorite now.

More bandwidth options that show the exact number will be perfect.

Let us know if you can implement this feature.

Thanks a lot, keep up the great work!

Best wishes from Türkiye!

Cornball5569, 2024-10-07:
Hello **Armel**, thank you for an advanced and constantly evolving firmware.

Further to original request from **xes**, three RF bandwidths are already defined in driver/BK4819.c;
 
BW_WIDE 25kHz,
BW_NARROW 12.5kHz 
BW_NARROWER 6.25kHz

The bandscope/spectrum analyser can utilise all three bandwidths (W/N/UN) in contrast to just the two utilised (W/N) by VFOs and channel memories. Would it be diffcult to add BW_NARROWER for normal operation of the the radio?

This would encompass the scenarios initally listed by **xes**.

25kHz for older wide FM services (some amateur, marine, GMRS,...)
12.5kHz for narrow FM serivces (amateur, PMR446, FRS, UHF CB,...)
6.25kHz for AM/USB services (airband, HF amateur, HF broadcast,...)

Best regards.

armel, 2024-10-07:
Hi @Cornball5569 

Thank you for your message.
I understand this request. But as it is, it would be complicated to implement. Bandwidth adjustment is done per channel. Offering 3 bandwidths instead of 2 would change a lot of things when saving channel settings. In short, it's not easy. 

Bests regards,

Armel F4HWN.

T1000-Cyberd, 2024-10-26:
Any update on  this? This is the last thing missing :)

armel, 2024-11-06:
v3.6 introduces a new feature (SetNFM) that lets you toggle between Narrow and Narrower. It's probably not exactly what you'd expect, but it's better than nothing.

closed 2024-11-06 by armel