armel/uv-k5-firmware-custom #101: Save bandwidth setting

xes, 2024-05-12:
Since now in vfo mode bandwidth is not saved and after a power cycle you loose it, would it be possible to fix this? 

No idea if this egzumer change on kamill repo could be of any help:
https://github.com/kamilsss655/uv-k5-firmware-custom/commit/ee871999db1f0d2e9c5ec898311442de747797d7

armel, 2024-05-12:
Hi,

Hummm, that doesn't seem right.
If you're in VFO mode and, for example, Wide, and you change to Narrow, this change will be reflected on restart. 
Or maybe I didn't understand. 

Armel F4HWN.

closed 2024-05-22 by armel

prokrypt, 2024-05-27:
I think he meant it's not saved if switched from the user-definable shortcuts.
Anyways, I consider it a *feature* rather than a bug. The most convenient method to change a setting should always be temporary.