armel/uv-k5-firmware-custom #102: RADIO FM BAND UE suggestion!

JoeLeonny, 2024-05-12:
Hi Armel, congratulations, I'm enjoying using your version 2.7 uk-uv firmware! I just wish that in the next version (pse) there would be two more functions available in the main menu:
Option in the FM radio menu for the European frequency range: 88 - 108 Mhz
By an option in the config. to unlock the frequency. from TX.
Thanks,
Kind regards

armel, 2024-05-14:
Hi,

Well, about unlock TX, why you don't use F Lock menu (64/71) available from the [hidden](https://github.com/armel/uv-k5-firmware-custom/wiki/Menu#hidden-menu) menu? 
About FM broadcast, I'm thinking of deleting this feature :)

73'

Armel F4HWN.

lapinos03, 2024-05-18:
> Option in the FM radio menu for the European frequency range: 88 - 108 Mhz

I dont't understand this request. There's already a 87.5-108MHz band. Do use F+1 keys to switch from one band to another.
Regards

armel, 2024-06-11:
In FM mode, use F+1 keys to switch from one band to another. I'm closing this issue.

closed 2024-06-11 by armel