armel/uv-k5-firmware-custom #24: Choix de capacité batterie menu BatType

LiNUT3k, 2024-05-12:
Bonjour,

Dans le menu BatTyp je n'ai le choix que de 1600 ou 2200 mah, or, j'ai acheté une batterie 3500 mah. Serait il possible d'ajouter cette option ? Merci

KDMcMullan, 2024-06-09:
In both cases, the relationship between charge and mV from 5% to 95% is linear. I feel like the battery capacity could be a scale factor. I need to think more about it. When the user enters a calibration value (hidden menu where we can enter a measure Vbatt), how is that value employed in the software?

prokrypt, 2024-06-09:
https://github.com/armel/uv-k5-firmware-custom/blob/9f585b8850888734640efaf18a3a25b661339412/helper/battery.c#L101C2-L101C68
`gBatteryVoltageAverage = (Voltage * 760) / gBatteryCalibration[3];` where gBatteryCalibration[3] is the value used to "calibrate"

armel, 2024-07-21:
I've just added support for 3500mAh batteries to v3.1.

closed 2024-07-21 by armel