armel/uv-k5-firmware-custom #29: Stuttering audio on NFM mode

dadanielat, 2024-05-12:
When received audio is reaching a certain loudness level, stuttering occurs when channel is set to NFM mode. Also happens with stations using legit devices that only support NFM. Seems to be a bug of egzumer 0.22, because in 0.21 it was ok.
I would suggest a quick fix of only sending in NFM and always receive in FM mode.
Was reported by many users in https://github.com/egzumer/uv-k5-firmware-custom/issues/467

IvanMaestroGTR, 2024-05-24:
> A way to reproduce might be a station with strong audio. Like I said, I have it when using my transceiver as crossband transponder. As you can see in the vid the stuttering is not on the (unfortunately not open) IJV firmware ( left) .
> 
>  20240423_214326_1.mp4

What if you try to reproduce this scenario but switch the dual watch into normal mode?

prokrypt, 2024-05-27:
I noticed it too on WIDE mode on the radio for some transmissions, BUT I discovered it also happens to a certain repeater (not the radio/firmware's fault!). I guess the audio overmodulation is wiping out or distorting the CTCSS tone...

IvanMaestroGTR, 2024-05-29:
I see i see
Usually in this case, most likely it would be ctcss or the calibration issue...
What about loading a generic or your saved calibration just to try out?

I had this issue back then when I was changing firmware, but reuploading calibration solved them.(Not sure your issue is same as mine)

dadanielat, 2024-05-29:
I don't use CTCSS and flashed directly from stock to this FW. Calibration data was not altered.

prokrypt, 2024-05-31:
Receiver in WIDE mode, no CTCSS. Baofeng UV-5R has no problems.
https://github.com/armel/uv-k5-firmware-custom/assets/1832555/7c87bd5b-3a58-4278-b431-99f32e515bea

prokrypt, 2024-05-31:
Interesting thing I just found in a test:
* transmitter in lowest power wide mode
* receiver in narrow mode
* receiver cuts in and out when I whistle into the transmitter
* when squelch is set to 9, it does NOT cut in and out... ?!?

I got bored of whistling and did not test further.