armel/uv-k5-firmware-custom #30: Using 999 channels instead of the original 200.

Midnight-Cowboy, 2024-05-12:
Hi Armel!
Using 999 channels instead of the original 200. However, this requires a rather complex hardware modification: a chip has to be desoldered and re-soldered.

This involves replacing the original 8K eeprom (24C64) with a new 128K one, such as 24M01.

closed 2024-05-12 by armel

xes, 2024-05-12:
....on the other side, there is someone that has already applied the eeprom mod and  now is blocked/limited and cannot change the firmware since only a limited set of firmwares support the new eeprom...

armel, 2024-05-12:
Everyone is free to update this EEPROM. But it's also up to you to accept the consequences. As I said, my firmware won't support this modification, unless Quansheng integrates it as standard. I can't take into account the hardware modifications suggested by one person or another. Last but not least, there is no memory space left at this time. 

But anyone is free to fork my firmware, remove certain features to free up memory and add support for this EEPROM, if they wish. My firmware is open source, that's the advantage :) 

Armel F4HWN.

xes, 2024-05-12:
I really appreciate your efforts and I understand your points, about device limits and also about development.
If you receive this kind of request it's just because your firmware is working really well and attracting users that until some time ago have used other firmwares, with - you know  - limits and advantages.

So... thank you for your efforts and patience!