armel/uv-k5-firmware-custom #31: Spectrum analyzer - Suggestion

OxQwerty, 2024-05-12:
Hello friend! 

Thanks for your firmware!
What to ask you, if it possible to add this very usefull feature:

https://github.com/kamilsss655/uv-k5-firmware-custom/wiki/40-%E2%80%90-Spectrum-analyzer#additions-extra-functionality

`Pressing PTT exits the spectrum and fine tuning screen and copy current peak frequency, modulation, step, bandwidth to VFO. Also entering spectrum will carry these settings from VFO (full integration). Now to enter fine tuning screen in spectrum press MENU. This allows you to save and respond to the frequencies found much faster.`

Anyway, thanks!

VA7FGT, 2024-05-14:
As an aside, Kamils fork has some insanely fast channel scanning, any chance of backporting that to your firmware or providing a build option for it? 

(As a side note, love that this work is under a proper FOSS license and everything you're doing, keep up the great stuff!)

mentanah, 2024-07-06:
Since there is now some space gained in FW 3.0 an additional enhancement for using the bandscope:
Is it possible to save the adjusted setting for re-entering the spectrum app? Until now, every time I toggle between VFO and bandscope I have to readjust my last settings and blacklist. When changeing the frequency scan range I could be resetted again. What dou you guys think?

goopaga, 2024-07-16:
> Since there is now some space gained in FW 3.0 an additional enhancement for using the bandscope 😉:
> Is it possible to save the adjusted setting for re-entering the spectrum app? Until now, every time I toggle between VFO and bandscope I have to readjust my last settings and blacklist. When changeing the frequency scan range it could be resetted again. What dou you guys think?

To me looks a great idea !

armel, 2024-07-17:
Don't dream, I have no plan of adding Kamils' improvements to this firmware. He certainly did a good job. But it would be a huge change with a lot of code modifications to add this to my firmware. 
What's more, it seems to me that the latest Kamils developments are not really public (only for [sponsors](https://github.com/kamilsss655/uv-k5-firmware-custom/wiki#sponsor-this-project)).