armel/uv-k5-firmware-custom #46: Screen garbling at 2m TX with full power

terminalvelocity23, 2024-05-12:
Hi Armel!

v 2.5 (and earlier versions as far as I can tell) display the following  behavior - when transmitting at 2m, the screen gets garbled, turned upside down or blanked out if you don't hold the radio in your hand and use One Push PTT. It reverts back to normal when you take the radio back in your hand in some cases, and always reverts back when you turn the transmission off. If you hold the radio in your hand, there's no problem. The exact frequencies depend on the antenna in question, but with the stock antenna I can reproduce it between 145 and 149 MHz.
Only transmitting at high power causes this effect, mid and low power modes doesn't seem to do that.
This doesn't happen with the stock firmware v.2.6 at the same frequencies, nor does it with IJV v.3, so this doesn't appear to me to be a hardware issue.
Both of my radios, UV5R Plus and K5(8) show the same behavior. 

I've tried reflashing the radios and resetting the settings, and this didn't help. Resetting the calibration to stock values didn't help, either.

Video's too big to attach it here, so here's [the cloud link](https://disk.yandex.ru/i/ceLvCueicicQ6w). And my CHIRP config file in the zip archive.

Thanks for your work!

[Quansheng_UV-K5 (egzumer + f4hwn)_20240415_new_squelch.zip](https://github.com/armel/uv-k5-firmware-custom-feat-F4HWN/files/14984307/Quansheng_UV-K5.egzumer.%2B.f4hwn._20240415_new_squelch.zip)

terminalvelocity23, 2024-05-23:
Turns out that switching the mic bar off helps with that, it's not 100% effective but makes it much better.

armel, 2024-05-23:
What are you talking about ?