armel/uv-k5-firmware-custom #62: S-meter wrong.

GonzoG, 2024-05-12:
S-meter calibration is still wrong. It's set to HF standard (S9 = -73dB), not VHF/UHF standard (S9 = -93dB, S0 = -140).
IARU-R1 VHF Handbook v.9.01, 2021:
_"Standardisation of S-meter readings
• One S-unit corresponds to a signal level difference of 6 dB,
• On the bands below 30 MHz a meter deviation of S-9 corresponds to an available power of -73
dBm from a continuous wave signal generator connected to the receiver input terminals,
• **On the bands above 30 MHz this available power shall be -93 dBm**"_

I know it can be changed with Chirp, but it should be correct in fw.

MatiasFernandez, 2025-03-04:
> First, hello and thank you for you message... Second, you're wrong. My S-meter is set to the VHF/UHF standard (S9 = -93dB, S0 = -140). My S-meter table is this one :
> 
> ```
> S9 + 40 dB	-53 dBm
> S9 + 30 dB	-63 dBm
> S9 + 20 dB	-73 dBm	
> S9 + 10 dB	-83 dBm
> 
> S9  -93 dBm	
> S8  -99 dBm
> S7 -105 dBm
> S6 -111 dBm
> S5 -117 dBm
> S4 -123 dBm
> S3 -129 dBm
> S2 -135 dBm
> S1 -141 dBm
> ```
> 
> If it's not the case on your K5 with my firmware, it's probably because your calibration eeprom is corrupt or was modified by an exotic firmware. That's not my problem and I can't control it. My firmware never overwrite calibration datas... [image]
> 
> Take a look at the photo in attachement : -93dbm on the TinySA output = -91dbm on the K5 (so around S9).
> 
> Hope this help.
> 
> Armel F4HWN.

Hello Armel, I'm new to the firmware and I'm trying to understand the s-meter settings. 

First of all, thanks for the firmware, it's fantastic.

I come from egzumer v0.22 and I'm not sure what's the current state of my calibrations.

Here you re saying that your s0 setting is at -140dbm but the table you shared has s1 as -141dbm. Shouldn't s0 have to be -147dbm to match the table you mentioned?

Thanks!

armel, 2025-03-04:
Hello,

S0 is out of the table and means nothing. 
Take a look [here](https://www.ok2kkw.com/more/s-meter.htm), for example. This corresponds to the table I used to calibrate the K5 Smeter with my firmware (for UHF/VHF bands). 

Armel, F4HWN.

MatiasFernandez, 2025-03-04:
Thanks! I thought that s0 was used by the firmware in combination with s9 to interpolate all the intermediate s-levels. So, as far as s9 is set to -93 (in CHIRP) the value for s0 doesn't matter?

MatiasFernandez, 2025-03-04:
Oh, sorry, I think I was not clear. I was referring to the "S-Meter S0 Level (dBm)" and "S-Meter S9 Level (dBm)" settings that appear in CHIRP

[image]

Now that I'm on my computer I checked your code and, as far as I'm seeing [here](https://github.com/armel/uv-k5-firmware-custom/blob/main/ui/main.c#L338), your firmware is actually hardcoding the values for s9 and s1 and completely ignoring `gEeprom.S0_LEVEL` and `gEeprom.S9_LEVEL` settings that I think are the ones I'm seeing in CHIRP and were introduced in the egzumer firmware.

[image]

[I opened a PR](https://github.com/armel/uv-k5-firmware-custom/pull/371) with some notes regarding this in the README file to make it a bit more explicit. I hope you find it helpful

armel, 2025-03-04:
Thank you 🙏🏻 Merged into armel:feature_update_v4 !