armel/uv-k5-firmware-custom #96: Sensibilité Squelch

F5GEB, 2024-05-12:
Bonjour, c'est une issue générale à tous les firmwares : Le seuil de déclenchement du squelch est réglé trop haut .
Est-il possible de changer  la valeur correspondante à squelch = 1 afin de permettre de déclencher sur des signaux plus faible ? En effet   des signaux  faibles sont parfaitement audible quand le squelch est ouvert manuellement (ou squelch = 0 sur des signaux aviation ou satcom), mais pas avec squelch = 1 merci et 73 de Eric F5GEB

KaSt, 2025-05-24:
Hi,
I came to this "issue" as was wondering why, on the same SQL setting of "1" my old Baofeng would make me hear something and my new Retevis RA-79 wouldn't. Consider I have absolutely no clue what I'm actually talking about, I'm a total noob so thought that no matter the manufacturer every SQL was "born the same".

As Armel suggested I first made a backup of the original values

[image]

Then proceeded changing the values for SQL 1 with something I could have found using a dice roll. Again, no clue what I was doing, just thought it was safe enough since I had a backup to go back. My original issue was that I was trying to listen to airband and nothing was ever "showing up", despite "monitor" would sometimes prove somebody was actually talking. Loads of static, but the voice was clear enough to understand what they were saying. So, proceeded with some values I drew from a hat like a magician draws a rabbit.

[image]

In my case, this actually "opened" the frequency enough to let me hear what people said. Maybe a coincidence, maybe... I don't know, but seems to have made a change. I don't suggest these values, make your own "guess". Stay safe, make a nice backup, try to adjust those values if needed.

KaSt, 2025-05-25:
I was celebrating too quickly, just a coincidence that signal managed to open predefined SQL-1 as in the original table.
I wanted to make some changes, trying other values and downloaded from radio. The values I found were the original ones, nothing gets saved there, could be a driver issue, I don't know.

armel, 2025-05-25:
Please note that you must check a box to specify that you want to upload calibration data.

[image]

After that, change Squelch values and upload to the radio...
I'm telling you, it works :)

Armel, F4HWN

armel, 2025-05-25:
Oui, il faut tester par itérations successives.
Il peut y avoir des variations d'un K5 à un autre.
À chacun de faire ses propres tests. Et au pire, remettre les valeurs initiales ;)
Bonne chance.

Armel, F4HWN