armel/uv-k5-firmware-custom #104: Suggestion réduction de la fonction Radio FM au minimum.

ic3fr, 2024-05-13:
Bonjour Armel,

En regardant le fonctionnement de la fonction radio FM (fm.c), je me posais cette question :  Pourrait on économiser de l espace mémoire en réduisant au strict minimum cette fonction ? C 'est à dire, juste pouvoir faire la recherche de la station en mode VFO, avec les flèches haut et bas.  Sachant que les stations sont différentes selon les Régions et que cette recherche est assez rapide . Eventuellement garder la dernière écoutée en VFO en mémoire ...

Je tenterais bien l expérience de mon coté si j y arrive ;p 

Encore merci pour ton excellent boulot et tes vidéos YT.
J'espère  être dans la bonne rubrique GITHUB .
73

Aymeric.

armel, 2024-05-14:
Bonjour Aymeric,

C'est une idée. En tout cas, ca se tente. Mais je pense que ca prendra tout de même beaucoup trop de place :)
Tiens moi au courant si tu tentes l'aventure.

73'

Armel F4HWN.

lapinos03, 2024-05-16:
Hello !
Bon, moi j'ai bossé un peu 💻 . Vu qu'il me manquait de la place en ROM et vu que je n'en ai rien à faire des mémorisations de stations FM, j'ai viré tout ça pour ne garder que le VFO et le Manual-SCAN. Exit aussi le changement de bande (F+1). Enfin, je veux les parties de code ont été désactivées par options de compilation (voir Makefile).
Je vous mets le commit en libre service si cela vous tente de tester. Comments are welcome, of course !
A+

https://github.com/lapinos03/uv-k5-firmware-custom/commit/48fe7ab87cd7397c43c0a823d387f338eb24903c

lapinos03, 2024-05-16:
```
arm-none-eabi-size f4hwn_simplefm
   text    data     bss     dec     hex filename
  60716      52    3140   63908    f9a4 f4hwn_simplefm
arm-none-eabi-size f4hwn
   text    data     bss     dec     hex filename
  61360      52    3148   64560    fc30 f4hwn
```

armel, 2024-05-16:
Bonsoir,

Oui, c'est une approche intéressante. Elle permet d'économiser 644 octets avec la solution que tu as retenue. J'étais arrivé globalement à quelque chose d'équivalent de mon coté en _éteignant_ certaines fonctionnalités dédiées à la broadcast FM.

Mais ca reste le fait de garder la prise en charge du BK1080, uniquement en charge de la broadcast FM, qui coûte toujours aussi cher. 

Actuellement, sur la v2.8 en préparation, j'ai ceci:

  ```
  text	   data	    bss	    dec	    hex	filename
  61168	     52	   3148	  64368	   fb70	f4hwn
```

Si je désactive le support de la broadcast FM, je tombe à ceci:

```
  text	   data	    bss	    dec	    hex	filename
  57064	     52	   3076	  60192	   eb20	f4hwn
```

Soit un gain de 4 104 octets !!!

Au passage, si je désactive le support du bande scope, j'obtiens ceci:

```
  text	   data	    bss	    dec	    hex	filename
  55912	     20	   2724	  58656	   e520	f4hwn
```

Soit un gain de 5 256 octets !!!

De ce que je perçois, j'ai l'impression que 80 à 90% des utilisateurs sont favorables à la suppression de la broadcast FM. En retour, j'ai remarqué que ceux qui veulent garder la broadcast FM n'utilisent pratiquement jamais (voir **vraiment** jamais) le bande scope.

Du coup, m'a position actuelle serait de partir sur l'approche suivante. Dès que le manque de place se fera ressentir (on s'en approche), je sortirais en parallèle 2 versions: 

- version A: avec le support de la broadcast FM mais sans le support du bande scope
- version B: avec le support du bande scope mais sans la broadcast FM

Ca laissera donc en gros 4 Ko de mémoire pour faire des choses et permettra de respirer un peu. Ce qui est, en l'état des choses, assez conséquent. 

Du reste, si vous avez un avis sur cette approche @ic3fr et @lapinos03 , je suis preneur évidement. 

73'

Armel F4HWN.

lapinos03, 2024-05-17:
Le _bande scope_ ? quèsaco ?

lapinos03, 2024-05-17:
Moi, je fais partie des 10% qui souhaite garder la radio FM. Cela permet d'écouter la musique/les infos quand on fait du bivouac en haute montagne, par exemple, plutôt que d'utiliser la 4G (s'il y a couverture) ce qui peut décharger significativement la batterie du téléphone...

ic3fr, 2024-05-17:
Bonjour,

Merci Lapinos, j'avais commencé à regarder comment réduire la FM, tu as été plus rapide que moi. 

Lapinos : F+5 le bande scope , pour détecter visuellement des réceptions sur une plage de fréquence

Oui Armel, après  avoir compilé sans la FM , j'avais aussi noté que le gain n était que de 4Ko. Je suis d'accord avec vous deux. Le Choix  de la version à utiliser serait le plus intéressant. (les FW de Matoz étaient sur le même principe.)
 
éventuellement :
A une version  avec  FM complète
B une version avec FM Minimum
C une version sans FM

Moi je suis comme Lapinos , L UVK5 est la radio que je sors partout , la possibilité d écouter les infos, et le PMR de secours est un plus. Quand on se déplace beaucoup Le VFO FM sans scan ni mémoire est grandement suffisant.

Par contre Armel, le firmware actuel est excellent, quelles autres fonctionnalités voudrais tu rajouter ? 

L'affichage du CW reçu direct sur l'écran :p ;p  :) . lol ?

Bon WE

73

lapinos03, 2024-05-17:
> Lapinos : F+5 le bande scope , pour détecter visuellement des réceptions sur une plage de fréquence

Ah ok ! Moi, j'appelle cela plutôt "analyseur de spectre (eng: spectrum analyzer)". Mais les youtubeurs semblent préférer _Band Scope_. Question de vocabulaire !

f1ixx, 2024-05-18:
Bonjour tout le monde

Le grain de sel de Boris ;-)
J'ai garde uniquement la saisie direct de la frequence FM (qui pourrait aussi etre
supprimer), la navigation UP/DOWN, gestion de la touche Exit (correction et retour).
J'arrive a ca :
<pre>arm-none-eabi-size f4hwn
   text	   data	    bss	    dec	    hex	filename
  59400	     52	   3100	  62552	   f458	f4hwn</pre>
Le scan, recheche et memoire => tous supprimer (enfin en ifdef).

Mais oui, le plus simple reste un <code>ENABLE_FMRADIO = 0</code>
Apres, j'ai d'autre modif, mais pas en rapport avec le sujet et qui gloutonne
nos precieux octets.

Bonne journee a tous
73's
Boris

armel, 2024-05-18:
@f1ixx L'échange peut être étendu aux autres modifs je pense :) Tu avais quoi comme autres pistes ?

73'

Armel F4HWN

f1ixx, 2024-05-18:
Non, pour la FM, juste cette piste pour gagne des octets. Garde le minimal.
Une autre idee que j'avais eu, mais pas encore regarde comment c'est faisable, etait d'ajouter
le mode WFM, que personne ne crie, je sais, j'ai regarde les datasheet.

Ce mode uniquement activable pour la bande du BK1080.
Au lieu d'envoyer sur le BK4819, on envoie sur le BK1080. On gagne pas de place niveau driver
mais peu etre un peu de gain sur la gestion (scan, saisie de frequence, ...). Mais peu etre aussi
pas la meilleur idee.

<Hors-Sujet>
j'ai brule des octets avec le screenshoot, le system que vous avez mis en place et ecrit un script
python pour les capture a la vole.
Repris le system de message de @joaquimorg mais avec support externe. J'ai supprimer toute la
couche GUI dans le talky pour utilise uniquement le port serie. A terme, je veux utiliser un telephone
avec bluetooth par exemple et je regards pour faire un system type packet avec indicatif et routage.

Bref ca bricole un peu histoire d'essayer des trucs et rien a voir avec ce fil ;-)

Question : y'a t'il un moyen d'avoir les previews de vos idees pour les firmwares?
Un emplacement pour echanger des idees, discute compilateur, modif de code
ou autre. Les issues ne sont peu etre pas le meilleur endroit.
</Hors-Sujet>

Un grand merci pour votre disponibilite et votre travaille

Boris

armel, 2024-05-18:
Ok pour l'ensemble @f1ixx c'est très intéressant.

Mon idée était aussi de basculer sur le BK1080, automatiquement, lorsqu'une QRG se trouvait sur le segment broadcast FM. Mais je n'ai pas encore réussi à tendre vers quelque chose de vraiment utilisable. Et ca coûte toujours assez cher coté occupation mémoire. Et puis surtout, ca veut dire travailler pour conserver le support de la broadcast FM dont, perso, je me fiche complètement. Et bosser sur un truc qui ne m'intéresse pas, ca me motive moyen...

C'est pour cela que je suis plutôt dans l'optique de produire, à terme, 2 versions: une avec la broadcast FM mais pas le bande scope. Et l'autre avec le bande scope, mais pas la broadcast FM. Ca reste simple à faire et ca libère carrément au moins 4 Ko de mémoire. 

Alors évidement, ca va tout de même coincer chez certains. En particulier celles et ceux qui ont un problème avec le simple fait de devoir faire un choix. Il y en a pour qui c'est quasi maladif. Devoir faire un choix est perçu comme un echec.

Mais d'un autre coté je ne suis pas magicien. Il est aujourd'hui impossible de faire évoluer ce firmware, sans faire de concession. Et je trouve dommage que l'évolution du firmware soit freiné par une fonctionnalité comme la broadcast FM qui n'a, de mon point de vue, rien à faire sur une poste dédié avant tout à de la communication radio. Ensuite, on parle d'un poste qui coûte entre 15 et 20€. A chacun de dédié un poste à l'écoute de la broadcast FM si besoin.

Mais ca, c'est mon avis en qualité de radio amateur, ce qui est loin d'être le cas de la majorité des utilisateurs de ce petit pocket :)

Pour l'endroit plus adapté, oui, il y a l'espace de discussion. Il est accessible [ici](https://github.com/armel/uv-k5-firmware-custom/discussions).

73'

Armel F4HWN.

Bouledogus, 2024-05-21:
Bonjour à tous,
Merci Armel pour ton firmware qui est pour moi est de très loin le plus abouti que j'ai pu tester.
je voulais juste ajouter un commentaire pour te dire que je fait parti de ceux qui n'utilise jamais le brodcast FM, je trouve ça absolument inutile pour un poste radioamateur.
Je suis impatient que tu supprime cette fonctionnalité au profit d'amélioration du band scope (que j'utilise beaucoup) ou l'ajout de la prise en charge d'autre modulations, je pense au dsb qui permet de faire du FT8 par exemple.

73 !
Arnaud - F4LBJ