armel/uv-k5-firmware-custom #108: Bug Raccourci Switch PTT

F4HWD, 2024-05-19:
Bonjour Armel.

Un nouveau bug à te signaler. J'ai affecté le F1 Long à Switch PTT. Ce raccourci fonctionne bien quansd le poste est en RxMode MAIN TX DUAL RX et DUAL RX RESPOND, mais pas en MAIN ONLY ni en CROSS BAND.

Si tu appuies sur le raccourci Switch PTT, le changement est bien pris en compte dans tous les cas, mais l'indicateur CL ou OP ne change pas dans les 2 RxMode cités plus haut. Il faut pour celà soit aller dans les menus, ou bien passer en émission pour que le changement s'affiche à l'écran.

Pas un bug critique, mais assez gênant quand même.

Bon courage encore un fois, et à bientôt.

73's de ON/F4HWD

armel, 2024-05-19:
Bonjour Maxime,

Désolé, mais je n'arrive pas à reproduire ce problème.
Ci dessous la vidéo pour confirmer.

https://github.com/armel/uv-k5-firmware-custom/assets/127889/13b39078-c388-4143-9fb6-0a60f0031bb1

73'

Armel F4HWN.

F4HWD, 2024-05-19:
Ah ? Bizarre car je viens de retester ce matin et le problème persiste sur mon talkie. Je t'ai fait une petite vidéo. A mon avis, le bug est dû à une autre option.

Si besoin, je peux t'envoyer en privé mon codeplug pour tenter de reproduire le bug.

https://github.com/armel/uv-k5-firmware-custom/assets/42820755/e345b9e6-2196-46d9-9030-40fb098908c9

armel, 2024-05-19:
J'aurais plusieurs remarques:

- déjà, j'ignore sur quelle version tu es, mais ca n'est pas la dernière stable, à savoir la v2.7. Pourquoi ? Tout simplement parce qu'à partir de la v2.7, la liste I et liste II, à droite du nom du canal mémoire, ne s'affiche plus comme ce que je vois sur ton écran (en chiffre romain). Donc si déjà je ne sais pas sur quelle version tu es, ca ne facilite pas la reproduction du problème. Et si tu me dis que tu es bien en v2.7, ca veut dire que tu as modifié des choses :) Et là, je ne répons plus de rien. 

- j'ai néanmoins tenté l'expérience avec une v2.6 et une v2.5, au cas ou. Je n'arrive toujours pas à reproduire le problème. J'ai mis les mêmes actions que toi, comme tu le montres en début de vidéo. 

- garde cependant en tête que, pour des raisons de temporisation, un appui LONG sur F1 prend **plus de temps** en mode XB et MO. On parle ici de _quelques dizaines de millisecondes_ hein (100ms = un battement de paupière). Ca n'est pas grand chose, mais ca fait la différence. Je te suggère d'insister sur l'appui LONG un peu plus longtemps dans ces 2 modes.

73'

Armel F4HWN.

F4HWD, 2024-05-19:
Bon je vais commencer tout de suite par la bonne nouvelle : problème résolu.

Par contre, je ne sais pas d'où venait le problème, car j'étais bien sur la dernière version (2.7). Je vais quand même donner la solution si jamais quelqu'un rencontre le même problème par la suite.

Un Reset All ne suffit pas. J'ai du flasher le firmware d'origine (penser à faire une copie de l'EEPROM avant). Une fois dedans j'ai fait un Reset All, puis j'ai flashé ton firmware. Après le redémarrage, encore un Reset All, et fini le bug d'affichage. Par contre il y a une petite subtilité, il ne faut **PAS** réinjecter l'ancien codeplug, sous peine de devoir recommencer toute la procédure. Il faut lire la radio, faire un copier-coller des canaux avec l'ancienne sauvegarde, et recopier à la main les paramètres, soit via ton driver Chirp, soit directement sur le poste. Et depuis, plus de problème. J'ai même retrouvé les nouveaux chiffres pour la ScanList, je ne les avais jamais vus, ce qui aurait du me mettre la puce à l'oreille. A noter aussi que sur le nouveau codeplug, certaines vieilles options ont disparu. A mon avis je devais traîner le bug depuis un bon moment.

Tout est bien qui finit bien, je clôture le dossier. Merci en tout cas pour ton aide, bon développement et à très vite.

73's de ON/F4HWD

closed 2024-05-19 by F4HWD