armel/uv-k5-firmware-custom #112: Radio Empfang

JustFun3D, 2024-05-23:
---------------------------------------------------------DE----------------------------------------------------------------
Ich habe 3 Geräte im Test

1. uv-k5 mit Si4732-A10 v1 FW 2.7
2. uv-k5 mit Si4732-A10 v2 FW 2.7
3. uv-k5 Original FW2.6

Bei keinem der Geräte habe ich Radio Empfang.
Wenn ich bei
1. uv-k5 mit Si4732-A10 v1 
2. uv-k5 mit Si4732-A10 v2  
die FW auf cec_0.51 Fullband installiere, funktioniert der Radi Empfang

Wenn ich alle 3 Geräte mit der Originalen Quansheng FW laufen lasse, habe ich auch kein Radio Empfang.

Antenne ist immer die Selbe.
---------------------------------------------------------EN----------------------------------------------------------------
I have 3 devices in the test 
1. uv-k5 with Si4732-A10 v1 FW 2.7 
3. uv-k5 with Si4732-A10 v2 FW 2.7 
3. uv-k5 original FW2.6

I don't have radio reception on any of the devices. If I install the FW on cec_0.51 full band on the 
1st uv-k5 with Si4732-A10 v1 
2nd uv-k5 with Si4732-A10 v2, 
the radio reception works. 

If I run all 3 devices with the original Quansheng FW, I have I also have no radio reception.   Antenna is always the same.

armel, 2024-05-23:
You asked a question [yesterday ](https://github.com/armel/uv-k5-firmware-custom/issues/111) and I answered it. My firmware **does not support the SI4732-A10** mod. So, please, don't try to use it with my firmware. It doesn't not work :)

JustFun3D, 2024-05-23:
It also doesn't work on a device that is original and has the 2.7 FW installed 😉

armel, 2024-05-23:
Sorry, I don't really understand the problem. 

My firmware is used all over the world, probably by thousands of people. And you think that, like you, they have no reception? :)

JustFun3D, 2024-05-24:
I don't think so, I just wrote what I found out 😉
If an SDR and other devices have reception on the same antenna, but the device with the 2.7FW doesn't 🤷
Devices with a different FW have reception on this antenna.

Your FW is absolutely first class, and I can live quite well without a radio, I didn't want to criticize you or your FW, I just wanted to point out something that I found out 😉

73

KDMcMullan, 2024-06-09:
If I understand correctly, none of your radios work with original Quansheng software.
I would suggest that your radios may have lost their calibration settings during reflashing.

closed 2024-06-24 by armel