armel/uv-k5-firmware-custom #113: Whizzing Sound During TX

misterbig, 2024-05-24:
Hello,
Thank you very much for your work!
Is it possible to remove the BLmax and BLmin in a future version?
The approximately 1kHz PWM can be heard during transmission.
Another possibility is to increase the PWM frequency to over 4kHz.
This problem has also been reported several times at EGZUMER
but not solved.
See here: https://github.com/egzumer/uv-k5-firmware-custom/issues/540
Thank you for your time and effort!
vy 73 dk1aj/Micha

Translated with DeepL.com (free version)

s1lviu, 2024-05-24:
Hope that this PR will solve the issue: https://github.com/armel/uv-k5-firmware-custom/pull/114 if @armel agrees. Thanks.

yo6nam, 2024-05-24:
I thought for a second that it can do FT8. So it's not a feature, it's a bug 🥇

misterbig, 2024-05-24:
Hello, 
by changing the PWM, the whistling has disappeared.
73 de dk1aj/Michael

armel, 2024-05-24:
Hello guys,

Thank you for this issue.

It's an interesting problem I've been working on. So I have a simple question: how old are you? :)

Why this question? Around 50% of people over 60 years old, lose their hearing in the frequencies between 3 kHz and 6 kHz. This is why increasing the PWM frequency from 1 kHz to 4 kHz works for some people. 

Given the age pyramid of amateur radio operators, increasing the PWM frequency, as you have done, may be a good idea. 

But for younger people, the noise will still be audible :)

73'

Armel F4HWN

s1lviu, 2024-05-24:
Hi, @armel 
Thanks for your interesting feedback. What about setting PWM around 25kHz?
Can be this a sweet spot for solving this problem in a much broader way?
Thanks!

misterbig, 2024-05-24:
I tested with 25kHz.
And that is far beyond the range of human hearing.
And also cannot be perceived as flickering of the display.
 It works perfectly.   73 de Michael/dk1aj

armel, 2024-05-24:
@s1lviu Ah ah,  I stopped hearing at 12 kHz :) 
I've seen your PR, thank you. I'll try to test at 25 kHz.

Armel F4HWN

s1lviu, 2024-05-25:
> @s1lviu Ah ah,  I stopped hearing at 12 kHz :) 
> 
> I've seen your PR, thank you. I'll try to test at 25 kHz.
> 
> 
> 
> Armel F4HWN

Great, @armel 
I also heard @misterbig talking with 25kHz variant and no tone were heard. Waiting for you feedback! ✌🏻
73 de YO6SAY

majonezz, 2024-05-26:
Just a note. When you rise a PWM frequency, backlight is getting dimmer at low backlight levels (1,2,3, etc). It can be compensated by rising PWM width probably.

closed 2024-05-26 by misterbig