armel/uv-k5-firmware-custom #115: Petit soucis avec Chrip pour le choix de la puissance Low + proposition pour nouveau dévelopement

ChrisWDC07, 2024-05-25:
Bonjour Armel, j'ai constaté (c'est peut-être pas toi qui gère cela) mais si tu peux transmettre à qui de droit.
Si l'on modifie la puissance Basse 20 mW 125 250 500 et 1 W via le menu de l'UVK5 l'affichage est correct soit L1, L2, L3, L4, L5
par contre si l'on modifie via CHIRP en reprogrammant le TX avec une des 5 valeurs proposée, le TX garde la valeur programmée par le Menu du TX et ne prends pas en compte celle programmée par Chirp. (bien sûr avec le denier module chirp chargé). Rien de grave bien sûr.

Le jour ou tu passes avec les 2 versions (FM broadcast et Band Scoop) je serait toujours intéressé et probablement pas le seul, d'avoir plus de 2 listes de scan... Cela permettrait de scanner différents type de fréquences, aviation PMR HAM Tour de France, Balise ou sonde météo etc...  
J'ai mis une étoile mais je sais pas si cela a fonctionné car je ne vois pas mon logging affiché au bout de la liste.
Merci à toi / F4WDC Christian

f1ixx, 2024-05-25:
Bonjour
Tester avec firmware 2.7 et driver chirp en ligne : tout semble fonctionner correctement chez moi.

Par contre, bien noter :
Lorsqu'on modifi un parametre dans la zone : F4HWN --- Beging setting, if ....
Il faut :
1. faire les modification sur le ou un des 9 parametres de cette zone 
2. Valider la checkBox tout en haut de la liste : Upload F4HWN setting to radio
3. Envoyer les parametre au poste (Upload to radio, comme d'habitude)
4. Decocher la checkbox
On continue et termine le parametrage de sa radio normalement.

En esperant avoir aide

73's
Boris

armel, 2024-05-25:
Bonjour et merci à @f1ixx pour son message.
Je n'aurais pas mieux répondu.
@ChrisWDC07 je te laisse contrôler à nouveau concernant Chirp.

Concernant l'augmentation du nombre de listes, j'y ai déjà pensé, mais ca change pas mal de choses au niveau du code, de la gestion des menus, etc. Néanmoins, je n'ai pas renoncé à me plonger sur la question en fonction de mon temps dispo évidement :)

73'

Armel F4HWN.

closed 2024-06-11 by armel