armel/uv-k5-firmware-custom #117: eeprom written to twice when manual scanning

prokrypt, 2024-05-25:
When scrolling/scanning through channels/frequencies, the eeprom is written to upon UP/DOWN buttonpress (when it initially switches to another channel/frequency), then it scans as normal while holding down the button. Then upon button release when it lands on the final channel/frequency, the eeprom is written to again. This happens on both memory mode and vfo, as well as FM broadcast mode. This bug doesn't exist in egzumer's firmware (I think), so maybe something was changed/removed/broken.

armel, 2024-05-25:
My firmware is more than 200 commits ahead of egzumer v0.22, so there are a lot of changes. For example, about the scan, since v2.4, a LONG press on MENU key, in * SCAN mode (List I or List II) temporarily exclude a memory channel (not working with * SCAN ALL). This may be the reason for this change...

prokrypt, 2024-05-25:
Oops sorry about that, false alert. I looked through the code and it seems it may have been present in egzumer's fw too. I tried my hand at fixing it (just simple removal of the bKeyHeld parts because it doesn't seem too reliable, and the code elsewhere still handles it correctly when button is released. I don't know what I'm doing but it seemed to work: https://github.com/prokrypt/uv-k5-firmware-custom/commit/33e9200a51e59b44c8dcb922aa5ddac68f2f7c0a

prokrypt, 2024-05-26:
Oops. After a few hours of testing, it looks like I broke some things...

prokrypt, 2024-05-26:
And all the things I broke (direct input of VFO/FM channels don't save until next keypress, VFO frequency no longer changes on single keypress, VFO direct input does nothing) are fixed... at the expense of more bytes used :melting_face:! https://github.com/prokrypt/uv-k5-firmware-custom/commit/b96138687f30e3c83bac45e6f533bc8978804944

prokrypt, 2024-05-26:
And after many more hours of testing, it seems like nothing broke this time!
Just ` && !bKeyPressed` added in two places. No more double eeprom writes. Yay.

prokrypt, 2024-05-27:
I noticed I only fixed the memory modes for VFOs and FM. Holding down up/down for frequency scanning still does a double eeprom write. I think I give up 🫠

prokrypt, 2024-07-01:
Finally! Found the last problem bit:
https://github.com/armel/uv-k5-firmware-custom/blob/15ff3d1512179e23ac3b7355228b0907e5304100/app/app.c#L2028
change to `if (!bKeyHeld && !bKeyPressed) {`
:tada:

armel, 2024-07-08:
There is now this test in your code `if ((!bKeyHeld && !bKeyPressed) || UI_MENU_GetCurrentMenuId()) {` ?

prokrypt, 2024-07-08:
Yes, that prevents the menu from closing when settings are changed

armel, 2024-07-08:
Ok, thank you.