armel/uv-k5-firmware-custom #118: PMR compile time flag request.

zardoz99, 2024-05-27:
Hi Armel, 

this is a very low priority request. Would it be possible for you to put a Makefile compile-time flag around your PMR additions? 

Your firmware is widely used outside of the European Union and the PMR code is only relevant to Europe, and not North America.

It would be nice to be able to exclude that code during a recompile of your firmware source, without having to manually remove all of the code segments.

The PMR (FRS/GMRS) frequencies in North America are based around 462 and 467 MHz, and not 446 MHz.

Regards,
Kelvin

armel, 2024-05-27:
Done in v2.8, still under development (take a look at this [commit](https://github.com/egzumer/uv-k5-firmware-custom/commit/5bae297e5f171caa0152eca7df98fe5d40da2028)).

Regards,

Armel F4HWN.

closed 2024-05-27 by armel

f1ixx, 2024-05-27:
hello
and why not make choice with compile with american PMR frequency ???
Maybe someone want have this option.

Change the PMR with PMR_FR, PMR_USA, PMG_?? ...
Not bigger size software, only option change.

73's
Boris

armel, 2024-05-27:
Hi Boris,

Free bands are very special for the USA (MURS and GMRS-FRS)

MURS : https://en.wikipedia.org/wiki/Multi-Use_Radio_Service
GMRS-FRS : https://en.wikipedia.org/wiki/General_Mobile_Radio_Service

But I can work on it.

73'

Armel F4HWN

armel, 2024-06-29:
I will add GMRS/FRS/MURS F_LOCK for North America zone in next release.

Armel F4HWN.