armel/uv-k5-firmware-custom #119: RX timer should only display/start when squelch is broken.

prokrypt, 2024-05-28:
Video should hopefully be self explanatory.
I have a tone set on a weather channel so it can be skipped during a scan. The RX toggling is fine as it shows dual watch is functioning correctly.

https://github.com/armel/uv-k5-firmware-custom/assets/1832555/f71dc0f2-0aab-4f45-9612-5a7c31fd35c3

prokrypt, 2024-07-02:
I guess it works as intended, but it's just doesn't feel very right...

Idea: reset timer only after a short period (200ms? 500ms?) so it doesn't keep resetting on a choppy signal. Sometimes when the radio/antenna is moved the signal can drop out for a moment.

armel, 2024-07-02:
I don't share your opinion here. 

This function is simple:

- if the radio receives (even a short signal), the RX counter starts,
- if the radio transmits, the TX counter starts. 

That's all. And it works decently enough. It lets you know, for example, how much time is left before the TOT, etc. 

Last but not least, note that there is no multitasking or real-time operation on this firmware. The scheduler is very basic and simple. 

In short, let's not waste time with these details, which in no way hinder the use of this radio in its primary functions: receiving and transmitting.

prokrypt, 2024-07-02:
But I think these details are what makes the firmware feel "premium".
Anyways, I added one tick (500ms) to the TX timer in convertTime() and removed one tick from RX, and it works almost perfectly now. The problem might be that the countdown timers are decremented too soon in the code or something :shrug:

closed 2024-07-08 by armel