armel/uv-k5-firmware-custom #121: feature request: channel input smarts

prokrypt, 2024-05-29:
When inputting a channel number and the input doesn't start with a 1 or 2, automatically add a 0 in front.
It could be done with 0 too, but it's nice to retain backwards compatibility with muscle memory... ¯\\\_(ツ)_/¯

f1ixx, 2024-06-01:
Hello

I read to make  a new entry system, but change is ... multi file, multi line and make some function change.
Not add big code, but modifications to long sections.

If you compile yourself, you can read [my repos](https://github.com/f1ixx/uv-k5-firmware-custom_Boris)
In Makefile active : ENABLE_MEMORY_KEYBOARD ?= 1

Entry procedure : to go on channel 12 : [1] [2] [MENU]
To entry in channel 5 : [5] [MENU]
to entry in chanel 145 : [1][4][5] (automatique change)

Note : my firmware is experimental and can't be used daily.. It's for my experimentation.
Possible add severial bug.
comment(s), idea(s), remark(s) or insult(s) are welcome

73's
Boris

f1ixx, 2024-06-04:
Hello

I'm make a text file to show my idea to change the function : [patch_entry_channel_memory.txt](https://github.com/user-attachments/files/15567543/patch_entry_channel_memory.txt)

In case anyone is interested and can explains my errors 

73's
Boris

KDMcMullan, 2024-06-09:
This suggestion could use better definition. For example, what if the selected channel is 30? We wouldn't want it to go to channel 3. One suggestion would be an entry timeout, like a TV remote: if you stop typing for 1 second, you're done?

prokrypt, 2024-06-10:
> For example, what if the selected channel is 30?

That would be [3] [0] [MENU], I'd assume.

f1ixx, 2024-06-14:
Hello

Timeout, why not. I read if i can't implante this.
I don't read the delay system for moment.

Yes, for chanel 30 : you push [3][0][menu]
For me, of course, [0][3][0] isn't a pretty solution.

73's

f1ixx, 2024-06-15:
hello

After 300 compilation, flash, tests et remake. I'm arrive a this solution.
I thinks in not very nice, but work. Of cause, make a modification add bug.
in misc.c :
<pre>change key_input_timeout_500ms          =  8000 / 500;  # 8 seconds
to : key_input_timeout_500ms          =  1000 / 500; # 1 second</pre>

in app/app.c :
line 1487
<pre>if (--gKeyInputCountdown == 0)
	{
        cancelUserInputModes();

	if (gBeepToPlay != BEEP_NONE)
	{
		AUDIO_PlayBeep(gBeepToPlay);
		gBeepToPlay = BEEP_NONE;
	}
...</pre>
change to :
<pre>if (--gKeyInputCountdown == 0)
	{
#ifdef ENABLE_MEMORY_KEYBOARD
      if (IS_MR_CHANNEL(gTxVfo->CHANNEL_SAVE))
        {
          MAIN_ProcessKeys(KEY_MENU, false, false);
          UI_DisplayMain(); // not logic but make job
          return;
        }
      else
#endif
        cancelUserInputModes();

 	if (gBeepToPlay != BEEP_NONE)
	{
		AUDIO_PlayBeep(gBeepToPlay);
		gBeepToPlay = BEEP_NONE;
	}
...</pre>

I thinks i forgot anything.
Adding this function is complicated for me with my poor programming knowledge.
it's not the best solution and I don't read the octet impact.

73's

armel, 2024-06-29:
Ok guys, I've implemented a solution that doesn't require you to press the M key (or another one). It seems to work not so bad. It will be enable in the next release. 

Armel.

prokrypt, 2024-06-29:
Uh, I don't know what is supposed to happen, exactly, but it's not working for me :sweat_smile:
It does nothing if I press less than 3 digits and let it expire.

[3][0][0] = Channel 44
[4][0][0] = 144
[5][0][0] = nothing
[6][0][0] = 88
[7][0][0] = 188
[8][0][0] = 32
[9][0][0] = 132

fascinating.

armel, 2024-06-29:
Just push 5 and you move to M5
Just push 17 and you move to M17
Just push 123 and you move to M123

Simple...

armel, 2024-06-29:
https://github.com/armel/uv-k5-firmware-custom/assets/127889/f4cd066f-36da-4faa-bd22-1d80ac3134e6

prokrypt, 2024-06-29:
Oops! Looks like I copied and pasted code wrong :sweat_smile:

But I also found a slight problem: it saves to eeprom on every keypress. Not very ideal...

prokrypt, 2024-06-29:
Furthermore, it saves the wrong channel when you don't input a complete (3 digits) channel.

Example:
* Press `2`. Let it change.
* Press `1`. Previous channel gets saved at this point. Channel changes to 1
* Power off and on radio
* Channel 2 is displayed instead of 1.

armel, 2024-06-29:
Ok, now it saves the good channel. This is fixed.
About eeprom, I will try to improve it, if I had time for that.
Life of eeprom does depend on temperature & time between writes. It's around 100 000 writes... So, personally, I wouldn't worry too much about it. By the way, it's easy to change eeprom and a K5 cost around 20 $ USD.

prokrypt, 2024-06-29:
Nice!

Indeed, the eeprom is pretty robust, but since it *can* die (I've heard of a few uv-5rs that no longer save settings) it's better to improve the code where we can. I think if you can make a system that saves all state settings after a period of no user input, that should extend the eeprom live considerably, and possibly save many bytes in the process.

armel, 2024-06-29:
Check the last [commit](https://github.com/egzumer/uv-k5-firmware-custom/commit/45987500f26e58f302ebb00ce5773ccef9b9c2a1) and try...

prokrypt, 2024-06-29:
I'll test it out when I receive my new "disposable" uv-k5 later today. I figured having two radios is easier to develop with (though I don't really know how to program. I just like to pretend I'm doing productive things).

prokrypt, 2024-06-30:
Just tried it out. Works great! :tada:

armel, 2024-07-02:
Fixed in next release.

closed 2024-07-02 by armel