armel/uv-k5-firmware-custom #122: Compander should not save upon selection

prokrypt, 2024-05-31:
Since the compander option is grouped with the other channel settings, it would be logical to assume that it does not save automagically.

In app/menu.c:
`SETTINGS_UpdateChannel(gTxVfo->CHANNEL_SAVE, gTxVfo, true, false, true);`
should be changed to
`SETTINGS_UpdateChannel(gTxVfo->CHANNEL_SAVE, gTxVfo, true, false, false);`

prokrypt, 2024-05-31:
Just a little quirk with the change I suggested. It seems the compander setting does not reset back to the original setting when you change channels and then go back. Looks like it may need some other change(s) to match the behavior of the other settings...

armel, 2024-06-25:
The real question would be whether the compander function is actually used by anyone. I never use it. And you ? 🤗

prokrypt, 2024-06-25:
I use it every now and then, which is how I found the "bug"... Although it's no big loss for me if it's gone, I know lots of radios (including very cheap ones) have this function, and it does make long distance communication more clear.

KDMcMullan, 2024-07-05:
I'd proposed removing it to free up program memory. This is a 5W HT, and I'd guess 80% of the user base are non-hams wanting to use it as a family walkie-talkie, or listen to airband. I respect that the three of us are hardly "representative", so I'm prepared to be corrected on this, but if anyone was taking the feature seriously, they'd be on here looking down the feature list and making suggestions.

armel, 2024-07-14:
Compander is an attribute specific to each channel. I won't change anything here 😌

closed 2024-07-14 by armel