armel/uv-k5-firmware-custom #123: bug lors du scan ?

stephane-rib, 2024-06-02:
Bonjour Armel,
j'ai remarqué un comportement bizarre lors du scan des mémoires (appui long sur *SCAN), en mode DW ou DWR (dual-watch) le scan se passe normalement en gardant le double affichage; en revanche en mode XB (cross-band) il passe automatiquement l'affichage en mode simple VFO, puis revient en double affichage lorsque le scan s'arrête, bug ou comportement normal?
Merci encore pour ton investissement
Stéphane

prokrypt, 2024-06-02:
I would think this is the correct behavior. If there were 2 VFOs on XB while scanning, it would either not work correctly or be extremely confusing.

armel, 2024-06-02:
Bonsoir Stéphane,

C'est tout à fait normal en mode Cross Band. 

73'

Armel F4HWN

stephane-rib, 2024-06-04:
merci
désolé pour le dérangement...

armel, 2024-06-04:
Personne n'a été dérangé :) 
Je clôture cette issue.

73'

Armel F4HWN

closed 2024-06-04 by armel