armel/uv-k5-firmware-custom #131: bug: backlight level 1 and 2 do not turn on backlight

prokrypt, 2024-06-14:
known issue when pwm frequency set too high. perhaps something can be adjusted to put the brightness scale back to normal.

armel, 2024-06-14:
I know, it's not perfect, but it will be hard to fix.

The choice is as follows: 

- low frequency for PWM = noise under modulation, but **there is a visible diff** between backlight levels 1 and 2 
- high frequency for PWM = no noise under modulation, but **there is no visible diff** between backlight levels 1 and 2 

The maximum frequency with a visible diff between backlight level 1 and 2 is around 7 kHz... But this frequency is audible (if your young...)

My choice is quickly made.

prokrypt, 2024-06-14:
Random idea: how about increasing the pwm frequency only when transmitting?
Also, have you tested other frequencies? 12kHz-18kHz might not be very audible over radio tx. 25kHz seems a bit overkill to me.

zardoz99, 2024-06-14:
> [Tracking image][https://github.com/notifications/beacon/AAQADJ5OE7EWUKKF5P4KANLZHJRP5A5CNFSM6AAAAABJJOZSLWWGG33NNVSW45C7OR4XAZNMJFZXG5LFINXW23LFNZ2KUY3PNVWWK3TUL5UWJTUBFQWCW.gif]

Would this not cause a visual problem if you were running at a low number in the brightness scale. When you transmitted, it would make the display dim and you might be led to believe that there was a battery problem. It might be confusing for users.
----------------------------------------

*From: *prokrypt ***@***.***>
*To: *armel/uv-k5-firmware-custom ***@***.***>
*CC: *Subscribed ***@***.***>
*Date: *Jun 13, 2024 11:46:09 p.m.
*Subject: *Re: [armel/uv-k5-firmware-custom] bug: backlight level 1 and 2 do not turn on backlight (Issue #131)

armel, 2024-06-14:
Agreed with @zardoz99 

Note that the human audible spectrum extends from 20 Hz to 20 kHz.
As I said, if the PWM frequency is above 7 kHz, there's no visible difference between backlight levels 1 and 2.
PWM frequency needs to be under 7kHz so... plain in human range.

Armel F4HWN.

prokrypt, 2024-06-14:
As of right now, there is no difference between levels 0, 1, and 2... I'm not sure if there is a hardware thing though as I have a KSUN UV60D. I might have some experimentation with it a little later on.

prokrypt, 2024-06-14:
I just did a simple test with a few different pwm values and an rtlsdr dongle and there was no difference in the squeal.
I'm not sure if the frequency is right as I haven't calibrated my SDR, but it seems there are two tones, one at ~3.750k and one at 7.5k. What is obvious though is that they remain relatively unchanged between different values. 
[image]

prokrypt, 2024-06-14:
And while I'm at it, I decided to test another uv60d with LOSEHU's firmware, and some cheap "frs" radio I had.
[image]

armel, 2024-07-02:
Fixed in next release.

closed 2024-07-02 by armel