armel/uv-k5-firmware-custom #132: [ENHANCEMENTS] FM function to demod (similar to IJV)

IvanMaestroGTR, 2024-06-14:
I was thinking of instead removing the whole fm function completely, is it possible to just port the fm function as a demodulator just like IJV to save space? So basically all the saved channels can be saved into the 200 memory bank. (Not sure if this idea is a little too late)

IvanMaestroGTR, 2024-06-14:
True...
If only the radio has more space for firmware (no more decidophobia)

armel, 2024-06-14:
I agree with you ;)

zardoz99, 2024-06-14:
> [Tracking image][https://github.com/notifications/beacon/AAQADJZI5H3Z23TVIAQ7LMDZHMJM3A5CNFSM6AAAAABJJXBU5WWGG33NNVSW45C7OR4XAZNMJFZXG5LFINXW23LFNZ2KUY3PNVWWK3TUL5UWJTUBHZP3W.gif]

The problem is, once you go beyond 64kB, the memory addressing becomes more complex. Within 64kB you can generate any memory address in just two bytes, without having to worry about bank switching or memory management devices. I wonder if they chose this particular chip for simplicity?
----------------------------------------

*From: *Armel FAUVEAU ***@***.***>
*To: *armel/uv-k5-firmware-custom ***@***.***>
*CC: *Subscribed ***@***.***>
*Date: *Jun 14, 2024 12:16:16 p.m.
*Subject: *Re: [armel/uv-k5-firmware-custom] [ENHANCEMENTS] FM function to demod (similar to IJV) (Issue #132)

armel, 2024-06-14:
Hey, the microcontroller inside your Quansheng K5 is not a Z80A or a 6502 :) You're not an 8-bit platform from the 80s :)

N32G030, embeded by the Quansheng K5, is a **32 bits** microcontroller ( based on Arm® Cortex®-M0).
There is no problem to deal with more than 64 KB of memory ;)

I contacted Quansheng at the end of March and suggest them to change it by, for example, an STM32F030CCT6. I'm pretty sure this STM32F030CCT6 is compatible with the current DP32G030 (who is a clone of STM32F030) used in the Quansheng K5. And this STM32F030CCT6 has 256 KB !!! It costs less than... 2 $ USD :)

Armel F4HWN.

IvanMaestroGTR, 2024-06-16:
Turns out actually there's a Green K6 that has the 512kb storage memory

Amazing stuff

prokrypt, 2024-06-16:
... but think of all the nice static resources we can store!
We can make more room for *all* the features! 😆
[image]

IvanMaestroGTR, 2024-06-17:
> As I've explained many times, don't confuse the EEPROM (which stores calibration data, settings and memory channels) with the microcontroller's flash memory (which stores the firmware) 😌 These are 2 different things. This Green K6, if it exists, does not allow you to install larger firmware.

Oops my bad... But honestly speaking, qs's website advertising on the green k5 basically talking about "bigger storage, allows you to put more features!". Was actually skeptical about this. But thanks for reminding me again!

IvanMaestroGTR, 2024-06-17:
> ... but think of all the nice static resources we can store! We can make more room for _all_ the features! 😆 ![IMG_20240612_162916687](https://private-user-images.githubusercontent.com/1832555/340116642-c4880470-4ba3-471e-9b0b-e8a44e907479.jpg?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MTg2MDQyNTAsIm5iZiI6MTcxODYwMzk1MCwicGF0aCI6Ii8xODMyNTU1LzM0MDExNjY0Mi1jNDg4MDQ3MC00YmEzLTQ3MWUtOWIwYi1lOGE0NGU5MDc0NzkuanBnP1gtQW16LUFsZ29yaXRobT1BV1M0LUhNQUMtU0hBMjU2JlgtQW16LUNyZWRlbnRpYWw9QUtJQVZDT0RZTFNBNTNQUUs0WkElMkYyMDI0MDYxNyUyRnVzLWVhc3QtMSUyRnMzJTJGYXdzNF9yZXF1ZXN0JlgtQW16LURhdGU9MjAyNDA2MTdUMDU1OTEwWiZYLUFtei1FeHBpcmVzPTMwMCZYLUFtei1TaWduYXR1cmU9NWQ0MDQ0Y2E3N2YzOGE3ODNhYWNhOGJmOTkzMTM0NjNkZTVhZDY5MjdmOGVhYzFkZmE0NWVlNDQ0MmZkMzZkMyZYLUFtei1TaWduZWRIZWFkZXJzPWhvc3QmYWN0b3JfaWQ9MCZrZXlfaWQ9MCZyZXBvX2lkPTAifQ.Nkb8jWXTf8Fknf0g9kXeqXCar8H4ARz1bOtUafpDTAI)
> 
> (okay I'm just only joking, slightly)

I'm not sure but there's quite a few dev (finger-countable) actually did use the static resources for some basic features. Not sure is it just me got that mistaken....

closed 2024-06-25 by armel

losehu, 2024-08-11:
> STM32F030CCT6

Can they be fully compatible?
I haven't reviewed the manual for this STM chip, so I'm not sure if their register addresses correspond. If feasible, I can try to experiment with it.

armel, 2024-08-11:
@losehu Probably not, even the number of pinouts differs. This is a real problem. After a lot of research on my side, I am now firmly convinced that evolving the DP32G030 microcontroller of the K5 to have more flash memory space and allow for larger firmware will be difficult. It would probably require changing a lot of things on the current PCB and probably starting development from scratch, which no developer will want to do.

losehu, 2024-08-11:
At present, I have implemented multi-system loading in the space of iic eeprom 512KB. I am thinking about whether there can be a virtual memory management mechanism to break through the limitation of firmware 60K.

---- Replied Message ----
| From | Armel ***@***.***> |
| Date | 08/11/2024 21:19 |
| To | armel/uv-k5-firmware-custom ***@***.***> |
| Cc | RUBO ***@***.***>,
Mention ***@***.***> |
| Subject | Re: [armel/uv-k5-firmware-custom] [ENHANCEMENTS] FM function to demod (similar to IJV) (Issue #132) |

@losehu Probably not, even the number of pinouts differs. This is a real problem. After a lot of research on my side, I am now firmly convinced that evolving the DP32G030 microcontroller of the K5 to have more flash memory space and allow for larger firmware will be difficult. It would probably require changing a lot of things on the current PCB and probably starting development from scratch, which no developer will want to do.

armel, 2024-08-11:
@losehu I know, you're doing some very impressive work on this K5 😌