armel/uv-k5-firmware-custom #133: How To Screenshot

KDMcMullan, 2024-06-14:
How do we use the screenshot feature, please? Thank you.

f1ixx, 2024-06-14:
Hello
If not make error (Armel confirm if correct)
1.Must compile with <code>ENABLE_FEAT_F4HWN_SCREENSHOT ?= 1</code>
2. Flash with "custom-custom" firmware
3. Remove function on M-Long (set to NONE)
4. When you navigate, you can take screenshoot with long press on [M]
 
You must have a serial monitoring and take screenshoot with plug  progamation  cable.<br>
Serial setting : 38400,8,N,1<br>

73's
Boris

KDMcMullan, 2024-06-14:
OKay, I think in v2.8, Armel said he'd enabled this. M long press still works for me, though. I'll hook u the serial line and try. Do you know the format of the serial data, please?

armel, 2024-06-14:
Hello, 

I confirm Boris @f1ixx answer.
This function is not activated by default and requires recompiling the firmware as Boris explains. 
About format, it's [PBM](https://en.wikipedia.org/wiki/Netpbm#PBM_example) P1.
Of course, after, it's possible to convert image in others formats with Imagick.

`convert Screenshot.ppm +level-colors Grey20,Orange1 -resize 256x192! target.png
`
Regards,

Armel F4HWN

f1ixx, 2024-06-14:
I've write a  dirty code python to display. You can read [here](https://github.com/f1ixx/F4HWN-Screenshoot/blob/main/screen_F4HWN.py)
It write for linux system. I can't test on window.
In code change "/dev/ttyUSB0" by "comx:" where x is serial com. number of you prog. cable.

Soft use : python 3.xx, tkinter, pyserial

No save function, but you can make screenshot.

73's

KDMcMullan, 2024-06-14:
Ah. I misread "not enabled by default" as the dormant code was already there, but in v2.8 you had enabled it. My mistake. Understood. Thank you.

armel, 2024-06-14:
You're welcome ;)

Armel F4HWN.

closed 2024-06-16 by KDMcMullan