armel/uv-k5-firmware-custom #134: Suggestion to Free Memory

KDMcMullan, 2024-06-14:
It strikes me that you have many great ideas or features and little space to implement them.

1) Is there anything to be gained by removing all but the fundamental Menu items, leaving the radio to largely be programmable by Chirp only? Frankly, programming channels on the HT itself is a nightmare anyway. As is modifying the scan lists.
2) Could you save a few bytes in code space by removing the roger bleeps? An unlicensed user won't be Txing (!) and a ham user is unlikely to want them.
3) Remove USB mode? Hmmm... Needs a bit more thought maybe.
4)  Fix the battery saver mode at maximum saving, remove the menu option.
5) Remove the option to switch OFF the AM fix. I think "off" is really only a test feature anyway.
6) I imagine the code to enter text is quite large. Power on message, channel name, etc ... if that code were removed, I wouldn't miss it.

Of course a lot of this is subjective and based on my use case, so not applicable to the many. However, in some cases, you and the wider community might agree!

Hope this is useful.

hypeARM, 2024-06-15:
I agree

armel, 2024-06-15:
Hello,

1 - I think it's important to keep a way of programming your radio from the radio itself, without having to use Chirp. This allows you to make small adjustments very quickly. And personally, I don't always have a PC with me. 

2 - If I remove Roger beep, it will be a revolution :) You wouldn't believe the number of messages I receive suggesting that I add others.

3 - Many users use USB as SWL.

4 - This point could be simplified, I agree.

5 - This point could be simplified too, I agree.

6 - Same opinion as for point 1.  It's important to keep a way of programming your radio from the radio itself.

Armel F4HWN.

KDMcMullan, 2024-06-16:
I imagine you've done more market research than me. I've never heard a Ham on 2m or 70cm using a roger bleep. I wonder if those requesting roger bleeps should be Txing at all! Yes, I recall custom bleeps being popular in the CB days!

What's you philosophy about messing around with the Menu index numbers? I don't thin it's too critical if the menu count deviates from the Egzumer.

Do you know if the Compander is a widely used feature?

hypeARM, 2024-06-16:
> I imagine you've done more market research than me. I've never heard a Ham on 2m or 70cm using a roger bleep. I wonder if those requesting roger bleeps should be Txing at all! Yes, I recall custom [...]

Roger beep is not well regarded here in Brazil, I have never used the compander function

I think the AM FIX function being removed from the menu is good, turning it off is very bad.

IvanMaestroGTR, 2024-06-16:
Plus I don't really think anyone will use AGC instead of the AM fix, so basically remove them is fine actually...

armel, 2024-06-23:
@KDMcMullan I don't know about Compander. It's a good question. I don't use it.

KDMcMullan, 2024-06-30:
It the bandwidth of a Tx channel is normally wide enough to incorporate all (or the most important) frequencies of human speech. The compander squishes the spoken frequencies down to a narrower range, then re-expands them on reception.
I certainly don't use this. It may well be a feature of the SDR hardware, in which case it only saves you a few bytes. 
Je crois que le "Roger Beep" n'est pas trop utilisé dans la communauté des radioamateurs: ça ne me fait rien sur 70cm/2m. Je peux comprendre, alors, qu'un bon % de votre audience ne soit pas des radioamateurs! ;-)

KDMcMullan, 2024-06-30:
Keypad lock is another feature which might not be heavily used.

armel, 2024-06-30:
The keyboard lock is used extensively by many users.

KDMcMullan, 2024-06-30:
Understood. Thought it was worth noting! :-)