armel/uv-k5-firmware-custom #135: Unable to flash via UVtools

tk5ep, 2024-06-15:
Hi,

When trying to flash via the suggested UVtools web page, I get the following error :

Connecting to the serial port...
Radio in flash mode detected.
Unusual error occured, check console for details.

Any guess on what's going wrong ? I can read the radio with Chirp, so connections are OK.

73, Patrick

armel, 2024-06-15:
Hi Patrick,

Which browser do you use?
Have you checked the console for details ?

Armel F4HWN.

tk5ep, 2024-06-15:
Hi Armel,
Thanks for the answer.
Strangely, back home I tried again and it worked flawlessly!

I should tried to reboot the computer before writing this issue. I did close/reload the UVtools page as well as the UV-K5, but this didn't help
Browser is Chrome.

Now, I've to get familiar with your SW. 

Sorry for the bandwidth.
73 Patrick

armel, 2024-06-15:
Hi Patrick,

No problem. I'm closing this issue.
Have fun with this firmware.

Armel F4HWN.

closed 2024-06-15 by armel