armel/uv-k5-firmware-custom #136: Firmware v2.8 - Comportement anormal du timer de réception

F6EGK, 2024-06-15:
Bonjour Armel,

Comme à l'habitude tu as encore fait du bon boulot, avec cette dernière version. Ci-après tu trouveras la description d'un bug rencontré ce jour.

Le timer de réception ne retourne plus à zéro, même en l'absence de signal (squelch actif). Le compteur s'incrémente en permanence, alors que celui-ci n'est plus visible à l'écran. Généralement après quelques minutes, le compteur saute (?) et se fige à 60:00. Un arrêt total du K5 et retrait de la batterie, font quelques fois repartir le compteur à zéro, mais c'est loin d'être systématique.

Le comportement est constaté pour tous les modes (AM, FM, USB), exploitation des mémoires ou VFO A/B. Une circonstance qui a peut-etre amené ce défaut à se manifester : en balade ce samedi 15, je me suis porté à l'écoute du contest 50 MHz, c'est à dire mode USB, squelch ouvert (MONITOR) ce qui a amené le compteur à démarrer et à s'incrémenter, et ce probablement pendant plus d'une heure. Je me contentais de modifier la fréquence de réception, à la recherche de stations, mais je n'ai hélas pas prêté attention à l'état du timer dans cette configuration.

Je n'ai pas reflashé le K5, dans l'hypothèse où en l'état tu souhaiterais que je fasse des manipulations complémentaires. Pour le moment il est bloqué à 60:00. Le timer émission (décompte) n'est par contre pas affecté. Cette problématique existait peut-être déjà sur la v2.7, mais je n'avais pas encore exploité le K5 dans la configuration décrite ci-dessus.

73 - Jean-Roger

PS : en discutant localement avec d'autres OM et suite à la modification de certains items, il apparaît que les nouvelles dénominations "Mode" et "Rx Mode" peuvent préter à confusion dans leur interprétation. Ce dernier item évoque davantage un mode de réception du type AM, FM ou USB qu'une configuration de réception (MO, DWR, DW, XB). Peut-être qu'un intitulé "Rx Conf" aiderait à une meilleure interprétation, et/ou remplacer "Mode" comme "Demod". De mon côté, j'opterai simplement pour remplacer "Rx Mode" par "Rx Conf" (je n'ai pas trouvé mieux !)

armel, 2024-06-15:
Bonsoir Jean-Roger,

Je ne suis pas sûr de tout comprendre.

Si le squelch est ouvert, la radio reçoit un signal. Donc le compteur en réception s'incrémente, ce qui est normal. Et je l'ai plafonné à 60 minutes. Donc, si le poste reste allumé avec un squelch ouvert, ce que tu as observé me semble normal. A vrai dire, je n'avais pas anticipé qu'on garde le squelch ouvert pendant aussi longtemps.

À noter que l'état du compteur ne persiste que dans la mémoire **volatile** du micro-contrôleur. Il n'est aucunement sauvegardé, par exemple, dans l'eeprom. Donc, un simple arret/redémarrage (inutile d'enlever la batterie) doit **forcément** le remettre à zéro. 

Bref, je ne comprends pas trop. Et je ne vois pas trop quoi faire ici. J'utilise ce timer depuis plus de 2 mois et je n'ai pas constaté de problème avec (autre que le problème fixé dans la v2.8, avec un compteur qui se remettait à zéro après un appui sur une touche du clavier pendant la réception d'un signal). 

Pour Mode versus RX Mode, c'est probablement l'appellation RX Mode qu'il conviendrait éventuellement de changer. J'ai reçu des tonnes de mails en provenance de France mais **surtout** du reste du monde, m'expliquant que Demodu n'était pas assez parlant et que Mode serait plus à propos. Ca me semblait tout à fait justifié. D'ou le changement.

73'

Armel F4HWN.

armel, 2024-06-15:
Bonsoir Jean-Roger,

Je viens de laisser mon K5, squelch ouvert, pendant 1 heure.
- le timer s'est normalement incrémenté,
- le timer s'est normalement arrêté à 60:00,
- le timer s'est normalement remis à zéro après extinction / allumage du poste.

Pour moi, tout est nominal.

73'

Armel F4HWN.

F6EGK, 2024-06-16:
Bonjour, et merci Armel pour tes informations.

J'ai réussi à reproduire le défaut, en procédant à la séquence suivante :

1. Squelch engagé, recherche à partir d'une position mémoire d'une station active (contrôle aérien de Reims dans mon cas)
--> le timer réception a un comportement normal : comptage, RAZ, comptage, etc.

2. Alors que le timer est dans une situation de comptage (station active), basculement sur une position mémoire inactive
--> si le squelch est maintenant ouvert (MONITOR), on fait le constat que le timer s'incrémente à partir de la dernière valeur de temps (pas de RAZ). L'alternat sur MONITOR ne change rien, l'incrémentation se poursuit . 

3. Toujours sur cette mémoire inactive, je coupe l'alimentation du K5 puis je la rétablis.
--> j'ouvre le squelch (MONITOR), le timer affiche 60:00 (l'interruption de l'alimentation puis le rétablissement ne changent rien à cet état).

[image]

4. Le timer ne retrouvera un fonctionnement normal sous réserve que l'on se porte à nouveau sur la réception d'une station active

Je sais, c'est ... tordu ! La version 2.7 n'a pas ce comportement. L'étape 2 se traduit toujours par une RAZ du timer à l'ouverture du squelch, bref tout est normal.

Si non, nous sommes bien d'accord que c'est l'item "Rx Mode" qu'il conviendrait peut-être de modifier.

73 - Jean-Roger

F6EGK, 2024-06-16:
Je comprends à la lecture de ton commentaire précédent, que ce que j'observe est probablement un effet de bord, entrainé par la correction apportée à la version 2.7  (RAZ timer suite à appui touche).

73 - Jean-Roger

f1ixx, 2024-06-16:
Bonjour Jean-Roger et Bonjour Armel

Bon, bien comme d'hab, je mets mon nez dans cette issue.
Alors, je m'embrouille avec les explications, ma petit tete a du mal.
Mais voila ce que je comprends. Si on ouvre en mode MONITOR, le compteur ne revient pas a zero.
J'ai donc ajoute dans le code du fichier : app/action.c
vers la ligne 151 :
<pre>APP_StartListening(FUNCTION_MONITOR);
		return;</pre>
devient :
<pre>APP_StartListening(FUNCTION_MONITOR);
    gRxTimerCountdown_500ms = 7200;
		return;</pre>
Surement que j'ai pas completement compris toutes les explications.
Mais lorsque on fait squelch ouvert/fermer, puis en ouverture du MONITOR, le compteur repart
bien de zero.

73's

Boris

armel, 2024-06-16:
Bonjour à tous les deux.

Ok, j'ai compris. J'avais joué avec le SQUELCH, en le baissant à zéro, plutôt que d'utiliser l'action MONITOR.
J'ai donc réussi à reproduire le bug.
Je viens de publier une version v2.8.1 qui, je l'espère, règle le problème.

73'

Armel F4HWN.

F6EGK, 2024-06-16:
Bonjour Armel, Boris,

Merci pour votre analyse et réactivité. Armel, je viens de pousser la v2.8.1, et là tout est OK.

Félicitations !

73 - Jean-Roger

armel, 2024-06-16:
Parfait, je cloture l'issue :) 
Excellente soirée à tous les 2.

Armel F4HWN.

closed 2024-06-16 by armel

f1ixx, 2024-06-16:
Bonsoir
Merci pour la reactivite, comme d'hab.

hors-sujet : commence a y'avoir plein de branche, mais je pull la feature_update ;-)

Bonne soiree a vous deux aussi
73's

Boris, F1IXX