armel/uv-k5-firmware-custom #138: After sending a string of DTMF using the * key, next keypress is eaten.

prokrypt, 2024-06-16:
Affects all keys except for PTT and channel buttons, both short press and long press.

armel, 2024-06-16:
Sorry, I don't understand where is the problem.

prokrypt, 2024-06-16:
Ah, while trying to make a video, I discovered the problem is a little more complex...
* Turn on radio
* Press "*"
* Input some DTMF
* Press PTT to send
* Press "*"
* [Optional] input DTMF
* Press PTT to send
* Next keypress (other than PTT or up/down) is ignored.

prokrypt, 2024-06-16:
https://github.com/armel/uv-k5-firmware-custom/assets/1832555/60689f07-5e63-466e-a546-56c0150af364

armel, 2024-06-16:
Ok.
I have no idea where this is coming from.
I tag the issue as a bug, if someone wants to take care of fixing it.

Armel F4HWN.

prokrypt, 2024-06-16:
https://github.com/armel/uv-k5-firmware-custom/assets/1832555/c8ee7caa-1ba9-4310-a5df-75fb8bc91b8e

getting closer to figuring it out... maybe

f1ixx, 2024-06-18:
Hello

I read and try to understand. Waw, high level for me.
But after many UART_printf to read data, may flash, i make this :
in file app/app.c :
line 1872 :
<pre>if (!bKeyPressed) {
	bFlag           = true;
	gPttWasReleased = false;
	}</pre>
change to
<pre>if (!bKeyPressed) {
	//bFlag           = true;
	gPttWasReleased = false;
	}</pre>
I make small test and it sames to work but but but as usual, fixes
a bug adds new bug(s).

If more harm than good, return to the origin and/or report the new problem(s)

73's
Boris

prokrypt, 2024-06-18:
@f1ixx getting very close! That does indeed fix the problem, but it is probably the wrong place to change code as it introduces a new bug:
* press "*" twice. oops!
* press "*" once then long press "*" again. hmmmm... interesting.

prokrypt, 2024-06-18:
I tried to have a go at it...
comment out line 681 (`gPttWasReleased = true;`) in main.c: https://github.com/armel/uv-k5-firmware-custom/blob/main/app/main.c#L681

It seems to work fine so far. Hmm.

prokrypt, 2024-06-25:
I've been testing this for the past few days, still no bugs! 🎉

armel, 2024-06-26:
Ok thank you 👍🏼 This issue will be fiexed in the next version.

closed 2024-06-26 by armel