armel/uv-k5-firmware-custom #142: bug: jumping to menu item via numbers do not work when ChName is selected

prokrypt, 2024-06-18:
Steps to reproduce:
* Press `MENU`
* Press `1` `7` to get to ChName
* Numbers (and `*` and `F`) do nothing. It thinks a channel name is being input, maybe.

This probably has something to do with `edit_index` in menu.c. Either it's not being set to -1, or something is changing it.
A fast fix would be to add a check for `gIsInSubMenu` in each of the affected key functions, but that's not ideal and takes up precious bytes...

prokrypt, 2024-06-18:
I fixed it by adding `edit_index = -1;` to app/menu.c, `MENU_ShowCurrentSetting` function, under `case MENU_MEM_NAME:`
(https://github.com/armel/uv-k5-firmware-custom/blob/main/app/menu.c#L1007C3)

Not sure if that's the correct place, but it's the best I could do...

prokrypt, 2024-06-18:
More changes: move `edit_index          = -1;` from https://github.com/armel/uv-k5-firmware-custom/blob/15ff3d1512179e23ac3b7355228b0907e5304100/app/menu.c#L1557C4-L1557C29 to https://github.com/armel/uv-k5-firmware-custom/blob/15ff3d1512179e23ac3b7355228b0907e5304100/app/menu.c#L1596

otherwise the same issue occurs when you keep pressing menu until the text input exits.

prokrypt, 2024-06-19:
Fixed it with a change in one spot only:
https://github.com/armel/uv-k5-firmware-custom/blob/15ff3d1512179e23ac3b7355228b0907e5304100/ui/menu.c#L746
change `if (!gIsInSubMenu || edit_index < 0)`
to
```
if (!gIsInSubMenu)
	edit_index = -1;
if (edit_index < 0)
```

armel, 2024-06-25:
Ok, thank you. It will be fixed in the next release.

closed 2024-06-25 by armel