armel/uv-k5-firmware-custom #144: A version for modded EEprom radios?

SutliffeProductions, 2024-06-20:
Hi. 
This is by far the most feature rich firmware I've seen so far really well done. Is there an possibility down the line, for a version which includes as many features as possible (pong, SMS, and custom boot images would be great) designed for radios with a larger eeprom mod installed? as it's a fairly easy mod to do, and would mean you are not as storange limited.

Keep up the good work!

prokrypt, 2024-06-20:
We will still be limited by 64KB flash, and it will take quite some work to get static resources moved to eeprom... Hardware mods are easy. The software part, not so much.

SutliffeProductions, 2024-06-20:
> We will still be limited by 64KB flash, and it will take quite some work to get static resources moved to eeprom... Hardware mods are easy. The software part, not so much.

I'm not sure what you mean, I could be wrong, but I was under the impression that the 64kb was the program EEprom space, as that is the only storage onboard. Apart from 8k of ram?

armel, 2024-06-20:
Hello @SutliffeProductions and @prokrypt :)

@prokrypt is right.

@SutliffeProductions, your Quansheng K5 embed a N32G030. It's  a 32 bits microcontroller (based on Arm® Cortex®-M0) with only 64 KB flash memory. 

As I've explained many times, don't confuse the EEPROM (which stores calibration data, settings and memory channels) with the microcontroller's flash memory (which stores the firmware) 😌 These are 2 different things. EEPROM is not (never) use to store or execute firmware. By the way, this EEPROM is very slow. 

In short, even if you update your EEPROM, you are still limited to 64 KB of flash memory, embedded in the microcontroller, to store your firmware.

73'

Armel F4HWN.

closed 2024-06-24 by armel

svh1985, 2025-09-22:
Maybe now it’s relevant, see post. The modder even used @armel’s firmware.

[https://makeprogress.ee/blog/from-feature-tetris-to-full-power-the-uv-k5-flex-pcb-brain-hack](url)