armel/uv-k5-firmware-custom #146: Feature: FM radio integrated in main list?

bartczar, 2024-06-23:
Hi

What I would like to have is FM broadcast stations names.

I know there are memory constraints but maybe it would be possible to remove current FM feature completely and integrate FM broadcast reception to main list - that would enable channel names automatically.

That would also fix automatically my other issue: sometimes I just want to listen to broadcast radio and quansheng switches to main VFO reception when there is something on. I could mark the channel as "main only" and problem solved.

Maybe FM button could be used to somehow limit the main list to channels from broadcast FM range, or introduce channel list 3 for this purpose?

Thanks!

armel, 2024-06-23:
Hi,

Frequency receiving has precedence over FM broadcast reception. FM will only work when there is no radio reception. That's why your Quansheng switches to main VFO reception when there is something on. 

Maybe, this could be improved. 

But I don't want to develop any new features about FM broadcast. FM broadcast mode is just an extra function, it works decently enough. Note that, your Quansheng is a HAM radio not a broadcast receiver. 

Armel F4HWN.

closed 2024-06-25 by armel

sergeibit, 2024-10-18:
Sorry asking this in a suggestion marked as won't fix, but I was about to ask the same thing.

Just to better understand: what prevents the radio from tuning and demodulating FM broadcast using the main VFO?

I know when the FM feature is enabled it's using another circuit and routines just for that.

But it's a hardware limitation? The main hardware can't tune and demodulate the FM broadcast band?

Or it's missing a software implementation? 

Again, sorry asking it here, I'm trying to understand better the radio hardware to start developing on it.

Thanks for the work you are doing on it!

armel, 2024-10-18:
Hi,

It's a hardware limitation. The BK4819 is not capable of demodulating wide FM (used by FM broadcasting). That's why your K5 uses a second chip : the BK1080. 

Armel.