armel/uv-k5-firmware-custom #149: feature request: direct toggling of scanlist numbers in MR mode

prokrypt, 2024-06-26:
when using the longpress 5 shortcut, it would be nice to take a direct number input to toggle that scanlist number instead of going through all the permutations.

would be extra handy with 3 scanlists :wink:

prokrypt, 2024-06-26:
an easy way to do this might be to hijack the Function option for 1(Band), 2(A/B) and 3(VFO MR). Since you already have differing functions when the function button is activated (compared to long-press), it shouldn't be too weird/difficult to get used to.

acbits, 2024-06-29:
> when using the longpress 5 shortcut, it would be nice to take a direct number input to toggle that scanlist number instead of going through all the permutations.
> 
> would be extra handy with 3 scanlists 😉

Do you mean long press 5 + <scan list number>? Potentially, 10 scan lists(0-9)? 

I do feel just 2 scan lists is not flexible.

armel, 2024-06-29:
10 lists... don't dream.

KDMcMullan, 2024-06-30:
I'm not sure I see the advantage in this feature request. At the moment, longpress 5 adds to scanlist 1, then you have 1&2, then 2, then off. It's no a million button-presses. Is this, in fact, a request for more scanlists?

armel, 2024-06-30:
Teasing : there will be a third list in the next version. 
Now a longpress 5 allows you to cycle between :

- list [0] (channels in no list)
- list [1] (channels in list 1)
- list [2] (channels in list 2)
- list [3] (channels in list 3)
- list [1, 2, 3] (channels in list 1 or 2 or 3)
- all (all channels...)

It works well enough as it is.

prokrypt, 2024-06-30:
But direct toggling is much easier~~

https://github.com/armel/uv-k5-firmware-custom/assets/1832555/bb91e668-b084-4c49-bda8-bde096ae9481

prokrypt, 2024-06-30:
Also, just an idea for direct selection of scanlists while scanning:

https://github.com/armel/uv-k5-firmware-custom/assets/1832555/e1f44d9a-d675-4700-9790-1f85f6616814

prokrypt, 2024-07-01:
Also, 10 scanlists may be too much, but I think 6 should be doable without too much trouble, maybe...

armel, 2024-07-08:
Added to v3.0...

closed 2024-07-08 by armel