armel/uv-k5-firmware-custom #155: Feature Request: Two "Calling" Channel Shortcuts

KDMcMullan, 2024-06-30:
When we long-press 9, we get out selected "calling channel". One suggestion is that we might be allowed to program two calling channels (one for each of the Tx bands). Long-press 9 might toggle between these two?
Of course some users might see "9" as an emergency feature, and might prefer only to have one...
It will be fun squeezing that into the 26 remaining free bytes of program memory!

armel, 2024-06-30:
Don't dream, with 26 remaining free bytes, it's impossible.
That's why, next release will be split in 2 versions :

Version 1 - with bandescope but without broadcast FM
Version 2 - with broadcast FM but without bandescope

prokrypt, 2024-06-30:
i just set all my "Calling" frequencies in one cluster of channels (channel 100, 101, 102, etc...) and have the call button change to 100, then use channel up button...

KDMcMullan, 2024-06-30:
Excellent. I'm v1 al the way! I was going to propose it, but I figured loads of people probably wanted it. That's the problem: the majority want EVERYTHING!
Good call splitting into two versions. Thank you.

KDMcMullan, 2024-06-30:
> i just set all my "Calling" frequencies in one cluster of channels (channel 100, 101, 102, etc...) and have the call button change to 100, then use channel up button...

Good hack. It makes perfect sense when someone says the words!!!

armel, 2024-07-02:
The solution proposed by @prokrypt does the job very well. I close this issue.

closed 2024-07-02 by armel