armel/uv-k5-firmware-custom #165: AM Fix Still in Chirp

KDMcMullan, 2024-07-05:
Is it normal that the AM fix can still be toggled in Chirp?
It implies that the code to handle it is still present; I'd have thought the "if" would have been deleted to save space.
Hope this is useful.

armel, 2024-07-05:
You are confusing several things: the firmware on one hand and the CHIRP driver on the other. Furthermore, I confirm that the code allowing the activation or deactivation of AM Fix support from the Quansheng menu has been removed. 

But, fortunately, the code that manages the improvement of AM demodulation is OBVIOUSLY still present. Otherwise, it would be a significant regression.

Finally, regarding CHIRP, you are looking at the wrong repository. So take a look on https://github.com/armel/uv-k5-chirp-driver

Armel F4HWN.

closed 2024-07-05 by armel

KDMcMullan, 2024-07-05:
I have the beta v3 Chirp driver loaded (https://github.com/armel/uv-k5-firmware-custom/releases/download/v3.0/f4hwn.chirp.driver.v3.0.beta.py) and am Iooking at checkbox labelled "AM Fix" (possibly generated by line 2283 of the same .PY file).

Yes, naturally, the code to actually cause the AM fix must still be present in code!

I had not looked at other repos for the Chirp driver source. Thank you. (I don't see the v3 beta there.)