armel/uv-k5-firmware-custom #170: Probleme avec les Mémoires avec chirp

Yves862, 2024-07-05:
J'ai utilisé la dernière version de chirp avec votre driver et ensuite j'ai mis le firmware f4hwn v3.0.
Le firmware est parfait, je l'adore.
Chirp vois bien le radio pour tout mais quand je transfert des frequences avec chirp rien ne reste dans la mémoire.
Mais à la main ça reste bien dans la mémoire.    Pouvez vous m'aider?

armel, 2024-07-05:
Quel version du driver avez-vous utilisé ?

Yves862, 2024-07-05:
La dernière version  beta

Yves862, 2024-07-05:
La dernière version bêta du driver
Et la version 3.0 avec bandscope.

J'ai passé de egzumer v 22 ou tout fonctionnait bien juste avant la mise a jour.

armel, 2024-07-05:
Aucun problème de mon coté.
Attendez éventuellement la version finale du driver chirp, quand elle sera prète.