armel/uv-k5-firmware-custom #173: Bug: Changing the memory channel while RX

mrkusypl, 2024-07-07:
Hi. While using the modification, I discovered a bug. The bug is related to the recent update to V3.0. If you try to change the channel in this way at the time of receiving, after changing the channel you continue to receive the previous one. I also noticed that interrupting the reception, for example, by pressing any button, causes a definite change of the channel to the desired one.

I show this in the attached video, when I receive the airport weather station and try to switch to PMR 3.

https://github.com/armel/uv-k5-firmware-custom/assets/110433144/7d148a33-2c49-4d7b-baa2-19e65202a368

armel, 2024-07-07:
Hummm, I'll have a look, but for the moment I can't reproduce the problem. 
I tested this:

Pocket A on TX (on PMR 3)
Pocket B on RX (on PMR 3)

Pocket B receives the signal. During reception on Pocket B, I moved to another channel (on PMR 4). Reception stopped.

prokrypt, 2024-07-07:
@armel did you try different methods of channel input? [0] [0] [4], [0] [4] (timeout), [4] (timeout)

prokrypt, 2024-07-07:
Steps to reproduce:
* go to any valid channel
* enable monitor mode
* press a number of a valid channel
* display updates to new channel but receiver is still open on previous channel

Does not happen when more than 1 digit is pressed.

mrkusypl, 2024-07-07:
It seems that the problem only occurs where you change the channel with a single digit, because two-digit or three-digit channels change the channel immediately when you enter the second and third digits.

mrkusypl, 2024-07-07:
It seems that while waiting for input the channels change like this:
- entering one digit (e.g. 1) does not change the channel, after timeout the channel changes to 1
- entering a second digit changes the channel to another using the first digit (e.g., entering 13 changes the channel to 1, after timeout the channel changes to 13)
- entering the third digit changes the channel to another using the first two digits, but immediately changes to the intended channel (e.g., entering 135 changes the channel to 13, but immediately changes to 135)

Here an example (Channel 1 is PMR 1, channel 13 is PMR 13, and channel 135 is CH-135):

https://github.com/armel/uv-k5-firmware-custom/assets/110433144/4fb1527f-4d0a-4875-8155-e98f9124ac7e

mrkusypl, 2024-07-07:
I decided to try to fix the error, and succeeded :) #175

armel, 2024-07-07:
Interesting bug

closed 2024-07-08 by armel

mrkusypl, 2024-07-14:
Here is a further extension of this fix: #178