armel/uv-k5-firmware-custom #174: Feature Request: MDC reception / Custom MDC transmission

KDMcMullan, 2024-07-07:
I don't know if the MDC ("Roger" menu option) is real MDC, or just a squawk that sounds like it. Nor do I even know if MDC is broadly used. As such this is a low priority feature suggestion if anyone can be bothered researching and implementing it!

It would be kind of cool if the HT could receive and decode and incoming MDC.

It would be even cooler if we had our own superset of MDC-1200, for example, which incorporated the callsign at the end of the transmission, which could be picked up and displayed by another enabled station.

armel, 2024-07-07:
Take a look at [OneOfEleven's](https://github.com/OneOfEleven/uv-k5-firmware-custom) firmware. Take a look to this LONG [issue](https://github.com/OneOfEleven/uv-k5-firmware-custom/issues/173) too.

Personally, I have no intention of improving Roger beep. By the way, mine is OFF (like 90% of users) :)

closed 2024-07-07 by armel

KDMcMullan, 2024-07-07:
I suggested removing the feature altogether, since IMO most ham users won't use it, and you indicated that many of the users wanted MORE Roger bleep options! I also have it "off".
Yes, I hadn't seen that it had been raised previously. Thank you.

armel, 2024-07-07:
I have no reason to delete it at the moment. It's factory firmware compliant. 
But I might delete it one day, if I need more space. That's not the case today.