armel/uv-k5-firmware-custom #176: Wide/Narrow Toggle during Reception

KDMcMullan, 2024-07-07:
Scenario:
- Key 2 (long press) is programmed to toggle wide/narrow.
- Mode is "Dual Rx" mode.
- VFO A is selected
- Incoming transmission on VFO B
- I wish to toggle wide/narrow on the station I'm listening to.

Problem:
- When I long press key 2, I can see the wide/narrow indicator toggle on VFO A (this is expected, since VFO A is selected).
- I can hear that it is the incoming audio on VFO B which is actually toggling between wide / and narrow.

Summary:
- The channel receiving audibly toggles Wide / Narrow, although it is the wide/narrow indicator on the primary VFO which visibly toggles.

KDMcMullan, 2024-07-07:
For info: can be replicated on v2.7 and v3.0.

armel, 2024-07-07:
Certainly, as is often the case, a bug that dates back to DualTachyon's work... Don't know how to fix it for the moment.

armel, 2024-07-13:
Fixed in next release.

closed 2024-07-13 by armel