armel/uv-k5-firmware-custom #177: Channel goes quiet if toggled TO during reception

KDMcMullan, 2024-07-07:
I'm on VFO A, but listening to a transmission on VFO B.
I toggle to VFO B, but the reception goes quiet. 
If I toggle back to VFO A, the reception on VFO B continues.

prokrypt, 2024-07-07:
Possible fix: https://github.com/prokrypt/uv-k5-firmware-custom/commit/6e7488efbc219d803ff7aa1457aff4068c7bfc5b

KDMcMullan, 2024-07-07:
I just noticed that this happens n v2.7. I cannot reproduce on v3.0.

armel, 2024-07-07:
Please don't post issue on old versions.
Issue can be reproduce on current version or not ?

KDMcMullan, 2024-07-07:
Of course, with my apologies. I thought this was on my v3.0 radio. It wasn't.

prokrypt, 2024-07-08:
I think this should be re-opened. I just tried it on 3.0 and it happens for me like 80%+ of the time:
* open same/different weather channels on vfo A and B
* switch between A and B
* opposite has RX (not always, but most of the time)

KDMcMullan, 2024-07-08:
At the time of creation, this issue had not been manifest on the latest version of firmware.
**It is actually reproducible on v3,** but not all the time.

armel, 2024-07-08:
I'm not able to reproduce it.

prokrypt, 2024-07-08:
Try this:
* set squelch to 0
* change VFO a/b
* the opposite VFO has RX

does not happen with monitor mode.

armel, 2024-07-20:
Again, I'm not able to reproduce it.

KDMcMullan, 2024-07-28:
I do agree that it's intermittent.