armel/uv-k5-firmware-custom #179: Spectrum analyzer - Suggestion II - loading memories

G8JNJ, 2024-07-09:
When the Spectrum Analyzer runs, and finds active frequencies, would it somehow be possible to perhaps automatically load them into some of the upper memories that could be reserved for the purpose ? I don't think most folks would use all 999 of them in normal use. Perhaps 991 to 999 would be a good range ?

It would then be possible to leave it running and determine local activity, then have the channels all ready to use in a memory location.

Thanks,

Martin

armel, 2024-07-09:
Hi Martin,

Well, 99% of users have only 200 channels (and an 8KB eeprom), not 999. And yes, I know that it's possible to change the eeprom and that Quansheng is starting to release an extended version with a 64KB eeprom (I even have 2 models in my possession sent precisely by Quansheng), but this configuration is still in the minority. If it becomes more _mainstream_, it's possible that I'll develop things like what you're proposing. But not right now. 

Armel F4HWN.

G8JNJ, 2024-07-09:
Hi Armel,

Sorry, yes you are correct, was thinking of the Baofeng UV-5RM.

Personally, I'd sacrifice a number of them to have this feature. 

Maybe let the user decide how many to set aside for the purpose.

Just a passing thought...

Thanks for the great Firmware, it's very impressive.

Regards,

Martin

armel, 2024-07-20:
Not planned for the moment.

closed 2024-07-20 by armel