armel/uv-k5-firmware-custom #180: Tx Lock - GMRS_FRS_MURS - Missing included frequencies

r4di8r, 2024-07-10:
RE: frequencies.c

Missing GMRS Repeater (+5) frequencies in frequencies.c , case F_LOCK_GMRS_FRS_MURS:

if (Frequency >= 46255000 && Frequency <= 46272500) // GMRS ONLY
				return 0;

The issue is that the Xmit frequencies for repeaters on 15R(467.550) and 22R(467.725) are not allowed.  The rest are overlapped / covered by "FRS Only"
	
if (Frequency >= 46756250 && Frequency <= 46771250) // FRS ONLY
				return 0;

GMRS repeater frequencies
Ch.     Xmit            Receive         GMRS Repeater Channel Description
15R     467.550         462.550         1
16R     467.575         462.575         2
17R     467.600         462.600         3
18R     467.625         462.625         4
19R     467.650         462.650         5
20R     467.675         462.675         6
21R     467.700         462.700         7
22R     467.725         462.725         8

armel, 2024-07-10:
Sorry, I'm not an expert in GMRS/FRS/MURS frequencies. 
Please feel free to send a PR (pull request) to correct this. 

Armel F4HWN.

r4di8r, 2024-07-10:
Unfortunately I'm unable to create PR, but this update will resolve the issue: (frequencies.c)  

	case F_LOCK_GMRS_FRS_MURS:
			// https://forums.radioreference.com/threads/the-great-unofficial-radioreference-frs-gmrs-murs-fact-sheet.275370/
			if (Frequency >= 46255000 && Frequency <= 46272500 ||
				Frequency >= 46755000 && Frequency <= 46772500) // FRS/GMRS
				return 0;
			if (Frequency == 15182000 || 
				Frequency == 15188000 || 
				Frequency == 15194000 || 
				Frequency == 15457000 || 
				Frequency == 15460000) // MURS
				return 0;
			break;

armel, 2024-07-10:
Thank you very much 👍🏼
This will be fixed in next release.

73'

Armel F4HWN

closed 2024-07-10 by armel