armel/uv-k5-firmware-custom #181: Channel memory when Exit is pressed

moreni57, 2024-07-10:
Hello with the original firmware of the UVK5 when we scan the memory channels and a communication occurs on the device when we press Exit it stops the scan and keeps the channel in question.

Now when we scan and press Exit, the scan stops but the channel which was for example speaking is not the same as the one where we stopped but it is the channel to which We had to start the scan before which was retained.

moreni57, 2024-07-10:
En français ma langue d'origine : 

Bonjour avec le firmware original du UVK5 dès que nous scannons les fréquences sauvegardé dans nos listes dès que une communication intervient nous appuyons sur Exit et le canal qui a parler en dernier restait en mémoire 

Hors avec votre Firmware dès qu'on effectue la même chose et bien le canal n'est pas gardé mais il revient sur celui avec lequel nous avons lancer le scan c'est un peu dérangeant.

prokrypt, 2024-07-10:
This is a feature, not a bug. Press M to stop the scan and return to the last active channel.

armel, 2024-07-10:
@moreni57 Après relecture du code et de la [documentation](https://github.com/armel/uv-k5-firmware-custom/wiki/Radio-operation#common-frequency--channel-scanning-features), je confirme la remarque de @prokrypt. La documentation précise que "_Le balayage peut être arrêté avec la touche EXIT, le résultat de la recherche sera ignoré et la fréquence / le canal reviendra à celui qui était réglé avant le début du balayage. Vous pouvez également arrêter le balayage avec la touche PTT ou MENU, auquel cas la fréquence / le canal sera réglé sur le dernier canal où une transmission a été trouvée._". Ce n'est donc pas un bug, c'est une fonctionnalité ;) 

@moreni57 After re-reading the code and the [documentation](https://github.com/armel/uv-k5-firmware-custom/wiki/Radio-operation#common-frequency--channel-scanning-features), I can confirm @prokrypt's comment. The documentation specifies that "_The scan can be stopped with the EXIT button, the search result will be ignored and frequency / channel will return to the one that was set before scan begun. Alternatively you can stop the scan with PTT or MENU button in which case the frequency / channel will be set to the last channel where transmission was found._". It's not a bug, it's a feature ;)

73'

Armel F4HWN.

closed 2024-07-10 by armel