armel/uv-k5-firmware-custom #189: WITHOUT VOX BUILD: F-LOCK RESET

Linden1, 2024-07-15:
With the WITHOUT VOX build installed F-LOCK will be reset on a reboot of the radio - both with the release and a build I compiled myself. May only be the UNLOCK ALL setting, I haven't tested further. I've gone back to the BANDSCOPE build and F_LOCK works as it should.

prokrypt, 2024-07-15:
It might only be a visual problem. Moving to another item and then back makes it display the right setting.

Linden1, 2024-07-15:
> It might only be a visual problem. Moving to another item and then back makes it display the right setting. Probably related to [egzumer#571](https://github.com/egzumer/uv-k5-firmware-custom/issues/571)

No. after reboot the setting is reset. This doesn't happen with the other builds, just the "without vox" build. Not related to the linked issue with egzumer.
BANDSCOPE version - works as it should.
BROADCAST version - works as it should.
WITHOUT VOX version - resets after a reboot.

Not a visual issue. Setting transfers to chirp.

armel, 2024-07-15:
Sorry. I'm not sure to understand where is the problem ? Using Voxless version...

1 -  I start my K5 by holding PTT + SIDE BUTTON 1️⃣ while turning on the radio and than Release All Keys.
2 - My K5 starts up and switches to menu 64/71 F Lock.
3 - I select (for example) PMR 446 (It was on CE HAM before...).
4 - I restart my K5 normaly... I could use RX and TX **only** on PMR 446 band  (outside this band, I have TX DISABLE).

So where is the problem ?

Armel.

armel, 2024-07-15:
By the way, I just tried to use the UNLOCK ALL option too. And it works like a charm.

Linden1, 2024-07-15:
That is indeed strange. I get the same problem with two other radios as well. These radios are freshly flashed and "Erase All" before accessing the hidden menu. I do the same method as you but select UNLOCK ALL. That works fine until I reboot the radio.
On my radios the F-LOCK is menu **65/72**.

armel, 2024-07-15:
F-LOCK is menu 64/71 on VoxLess version (and 65/72 on others... depending if Vox menu 52/64 is present or not...).

Armel.

Linden1, 2024-07-15:
You are indeed correct. I've downloaded the settings back to chirp and it's fully unlocked. Can transmit out of band should I want to.
You can close this - it's a non issue. Never noticed these visual bugs with the Bandscope or the Broadcast version, so I panicked myself.

closed 2024-07-15 by Linden1