armel/uv-k5-firmware-custom #190: bursts of noise on spectrum with memory name display feature

prokrypt, 2024-07-15:
I think it happens when it scans through all the memory channels on each screen redraw. I suggest you cache the name and rescan only when the frequency has changed.

https://github.com/user-attachments/assets/fba70a4d-101b-413d-9d6d-992674966aed

armel, 2024-07-15:
This noise is certainly not linked to the display of the channel name on the screen, but more likely to a surrounding AND periodic disturbance.

prokrypt, 2024-07-15:
It only happens on the firmware with the FM feature disabled.

prokrypt, 2024-07-15:
Example of the input lag:

https://github.com/user-attachments/assets/1026fe29-57fa-4016-ace0-217b8953cdb8

prokrypt, 2024-07-16:
I fixed the problem, but don't know if my method is ideal. I'll leave it to you to implement whichever way you like:
[image]

prokrypt, 2024-07-24:
It only works on the non-fm-broadcast firmware. On my own voxless fork, I have it display only an "M" to save space from looking up channel names.

jepotter224, 2024-08-21:
I have the voxless v 3.2 loaded and I never see an M with a memory channel is hit.....not sure why?

armel, 2024-08-21:
And of course, you did read the description? About VoxLess version : VOX, Audio bare, Air Copy support and **new bandscope evolutions since v3.0 were been removed**...
If you want this feature, use bandscope version :)

jepotter224, 2024-08-21:
Thank you for the clarification and direction for finding this feature......FYI - appreciate all you work!

armel, 2024-08-21:
@jepotter224 you're welcome 🤗

closed 2024-10-15 by armel