armel/uv-k5-firmware-custom #193: List scanning

Thijss81, 2024-07-17:
List scanning. For example set scanlist2 to start on channel 50. After sure? And menu to confirm. It's sets itself back to the setting it was. Same for Slist1 and Slist3? 

Hope you can fix this.

Thnx!

armel, 2024-07-18:
Sorry, I don't understand.
SList1 is used to list channels assigned to scan list 1.
SList2 is used to list channels assigned to scan list 2.
And SList3 is used to list channels assigned to scan list 3.

That's all.

armel, 2024-07-22:
No news... I close this issue.

closed 2024-07-22 by armel

Thijss81, 2024-07-27:
Sorry for my late reaction.

My sincere apologies! I will first do a better job of figuring things out instead of claiming unjustified problems. I enjoy your firmware, thank you! 

Have a nice day

With regards,

Matthijs

commit f8ff71aa (2024-01-14): Fix #193: DTMF interrupt temporarily disabled after TX causing DTMF tones lost