armel/uv-k5-firmware-custom #195: F_LCK "disable all" as additional menu option in the main menu to keep it saved over chirps

rob-w, 2024-07-19:
To have the device as scanner only , it would be nice to have a way to disable all TX from the normal menu and keep that saved ( also when using chirp).
Maybe introduce a separate option under the normal menu ?

Currently this feature is only reachable via the extended menu and gets reset ( at least for me) when downloading a chirp.

rob-w, 2024-07-19:
i just started today playing with my new toy and this fascinating code. keep up the good work.

armel, 2024-07-19:
According to me, F_LOCK settings is save by Chirp (as others settings).
Look at Preferences tab, Unlock settings option.

[image]

rob-w, 2024-07-20:
But when i try to alter this option in chirp i get a hint that i hve to do that inside the device ?! Did i miss something in chirp ?

armel, 2024-07-20:
Only for UNLOCK_ALL option... 
I don't know if you're an amateur radio operator or if you have a radio operator's certificate. But remember that while listening to frequencies is open for all, transmitting is not. 
Take a look to the [documentation](https://github.com/armel/uv-k5-firmware-custom/wiki/Radio-operation#how-to-unlock-tx-on-all-bands). 

Armel F4HWN

closed 2024-07-20 by armel

rob-w, 2024-07-20:
My interest is pure into listening , hence my request for the feature to be reachable from the normal menu and staying over chirps uploads. Just to clarify , i dont want to have "unlock all" in this feature rather the opposite , "disable all" . As i am new to all this, maybe i didnt get it all correct in the usage of chirp and so ... let me double check and get back to you ... maybe my report comes down to a bug which is , chirp is overwriting my in-phone setting i made to "disable_all TX " when i upload a new chirp config

armel, 2024-07-20:
As I said, setting F_LOCK directly from the transceiver (via the hidden menu) or from Chirp works. Everyone has been using it this way for months. By the way, if you think that it's a Chirp issue, post an issue on [Chirp repository](https://github.com/armel/uv-k5-chirp-driver/issues) ;) 

Armel F4HWN

rob-w, 2024-07-20:
ok, sorry for all the fuzz
i used a outdated/mismatching chirp module. All works as expected now! Thx & Sorry , will keep that in mind and those 2 projects insync.

armel, 2024-07-20:
You're welcome ;)