armel/uv-k5-firmware-custom #206: Second Power Menu

marmatBCN, 2024-07-23:
Hi Armel
I don't understand the principle of the 2 Power Menu...? in both I can set any power value in the first there is only an additional USER option.
Can you explain it?

prokrypt, 2024-07-23:
I think the USER option may be for backwards compatibility with previous firmwares. If you have already set power levels for each channel individually, you can ignore it. Otherwise there is another menu item to change the USER power level.

armel, 2024-07-23:
I explained all of this in my last video.

Before v3.1, with the Power menu (02/64), you can set a channel's power to LOW, MID and HIGH.Subsequently, the SetLow menu (56/64) was used to set LOW power and choose between LOW1 and LOW5. But when you made a change via the SteLow menu, ALL THE CHANNELS that were at LOW power were affected.  

Since v3.1, with the Power menu (02/64), you can set a channel's power to LOW1, LOW2, LOW3, LOW4, LOW5, MID and HIGH. This was not possible before. This means that channel M01 can be set to LOW1, channel M02 to LOW2, channel M10 to LOW 4, and so on. 

In addition, via the Power menu (02/64) you can also select USER. If you choose USER, the power level will be that selected in the SetPwr menu (56/64). And as with SetLow (now replaced by SetPwr), if you modify the power from SetPwr, ALL THE CHANNELS that were at USER power were affected. 

Regards, 

Armel F4HWN.

marmatBCN, 2024-07-24:
Thank you now all is clear

armel, 2024-07-24:
Ok. I'm closing this issue 👍🏼

closed 2024-07-24 by armel