armel/uv-k5-firmware-custom #208: Feature Request

rob-spotter, 2024-07-24:
Nice work on this Firmware. Would it be possible to add a feature like the IJV FW when you hold down the A/B button for an extended amount of time to switch the current selected channel into single channel mode vs duel mode then back to duel mode if in single channel mode with the same A/B extended button press sequence. I love this FW, but going to the menu to do it, is kind of a pain. It would be a great addition if it can be done easily. Thank you so much! 😊

armel, 2024-07-24:
Thank you.

Take a look at MAIN ONLY action. That's exactly what this action does. And you can assign it to the F1 (LONG or SHORT), F2 (LONG or SHORT) or M (LONG) key.

closed 2024-07-24 by rob-spotter

armel, 2024-07-24:
Those programable button settings ARE in the Chirp driver :)
[image]