armel/uv-k5-firmware-custom #211: Feature request: Improved exclusion of channels from scanning

mrkusypl, 2024-07-24:
When using radio I decided to scan the channels in the 0 list, but I came across a channel with interference so I decided to exclude it from the scan. The problem is that instead of excluding, the receive modulation changes. The same thing happens when scanning all channels.

It would be a good idea to add channel exclusion when scanning all channels or channels in list 0.

Also I would suggest adding an icon to indicate that channels have been excluded from scanning.

prokrypt, 2024-07-24:
The way the skip code works is to remove the channel from the list (or rather, it removes the list from the channel!), so it doesn't currently work with anything other than single lists (1, 2, and 3).

One way to solve this issue may be to make a global skip list that applies to all scan modes. Doing so will also make it simple to display a skip indicator. Maybe we can cancel the skip with Exit (short or long press?) on the main screen.

armel, 2024-07-24:
Excluding a channel from the scan list only works if you scan lists [1], [2], [3], or [1, 2, 3]. However, it does not work with lists [0] and [ALL]. The reason is simple: excluding a memory channel means removing it from the list it belongs to. And when a channel does not belong to any list... it joins list [0]. So if you scan list [0], excluding a channel would mean putting it in list [0], which makes no sense.

Due to the lack of memory at the time of developing this feature, I couldn't do better. But I could try to improve this, if I have time. 

I flag this issue as enhancement.

prokrypt, 2024-07-24:
I just tried a quick and ugly hack and it seemed to work... HMMMmmm
[image]

armel, 2024-07-25:
Ok @mrkusypl.
In the next release, it will be possible to add channel exclusion when scanning all channels or channels in list 0. And there will be a small E (inverted video) instead of the numbers of lists (inverted video) to indicate that the channel has been excluded from scanning.

I close this issue.

closed 2024-07-25 by armel

prokrypt, 2024-07-25:
On mine, I have long-press exit remove the channel skip.
[image]

mrkusypl, 2024-07-25:
@armel and @prokrypt,

I decided to try to implement channel exclusion. So I created a variable of 25 bytes (200 bits) where each bit corresponds to each channel. A bit set to ‘1’ indicates that the channel is to be excluded during the scan. Holding down the EXIT button for longer resets the list.

Although I'm happy with the result, I feel a bit unsatisfied with the amount of memory -* 72 bytes* is a bit much.

I also tried implementing an ‘E’ icon that displays next to excluded channels. This looks good, however it increases the code by another *76 bytes*!

Modified code: 0b71cbb526a298b04107fef0b98c32a923f5b6e1 and below the picture with the excluded channel:

[image]

prokrypt, 2024-07-25:
functions use up quite a bit of space. better to not use them even if you have to use duplicate code in a few places.

armel, 2024-07-25:
@mrkusypl I used the same approach as you, I think. Take a look at the [feature_v3 branch](https://github.com/armel/uv-k5-firmware-custom/commits/feature_update_v3/) if necessary. 

By the way, v3.2 still in development, with this improvement regarding exclusion, is smaller in size than v3.1 (around 80 bytes). 

But you know, I don't really care about memory size at this stage, anyway. There's a time for testing things, developing new features and improving others. And there's a time for refactoring code and optimizing what can be optimized. 

Armel F4HWN.

armel, 2024-07-25:
@prokrypt How does it work if you set ChDisp (35/64) to FREQ?

prokrypt, 2024-07-25:
> @prokrypt How does it work if you set ChDisp (35/64) to FREQ?

badly 😂😂😂
oops

armel, 2024-07-25:
@prokrypt  Yes, it's a solution 🤭 But large-digit displays are used a lot. Believe me, the average age of amateur radio operators is quite old. One of the first requests I received, earlier this year, was to develop the Main Only mode precisely for this reason 😌

mrkusypl, 2024-07-25:
> @mrkusypl I used the same approach as you, I think. Take a look at the [feature_v3 branch](https://github.com/armel/uv-k5-firmware-custom/commits/feature_update_v3/) if necessary.

Huh, I didn't even notice that 🤨

I checked it out, and it looks much better than my work. Well done!

mrkusypl, 2024-07-26:
One more note on the exclusion with the ‘ALL’ option, as the exclusion with this option does not work. All I had to do was change one line in chFrScanner.c (line 282)

from:
```c
chan = RADIO_FindNextChannel(gNextMrChannel + gScanStateDir, gScanStateDir, (gEeprom.SCAN_LIST_DEFAULT < 5) ? true : false, gEeprom.SCAN_LIST_DEFAULT);
``` 

to:
```c
chan = RADIO_FindNextChannel(gNextMrChannel + gScanStateDir, gScanStateDir, true, gEeprom.SCAN_LIST_DEFAULT);
```

It also saved 16 bytes!

armel, 2024-07-26:
Oops, my bad 🤭 Fixed. Thank you mate 🤗
And yes, 16 bytes free !

Armel F4HWN.