armel/uv-k5-firmware-custom #213: Toggle channel exclusion outside of scanning

mrkusypl, 2024-07-26:
I came up with the idea of being able to exclude channels from scanning outside of scanning mode. There is exclusion removal, but the option to add exclusion outside of scanning would be a good idea.

However, in order to be able to implement, I needed to move the exclusion from the MENU button to EXIT, where there is no action under the long button (not counting exiting DTMF input).

Along with changing the exclusion button to EXIT, I had to change the function from the EXIT button so that holding the button down for a long time while scanning does not exit scan mode (something like the MENU button, which changes modulation when held down for a long time, but does not enter the menu).

This also managed to save **64 bytes**! :)

Modified code: 6beed07fa1234db2df6b27f0c7a6bd1dd8acd59

prokrypt, 2024-07-26:
Personally I don't think being able to exclude a channel from the main screen has much use, but I had implemented something a bit similar in my fork:
monitor mode can be enabled while scan is running, then you can use up/down to go through the scanlist. holding down M will add it to exclude list.
my target feature was letting monitor mode remain enabled in scan mode. the ability to exclude like that was only a byproduct.

mrkusypl, 2024-07-27:
@prokrypt, I was thinking of being able to exclude channels from the main mode so that I don't receive them when scanning. You won't have to wait for the RX of these channels to exclude them, you can just do it manually from main mode.

armel, 2024-07-27:
While the ability to exclude a channel during a scan seems important to me, excluding a channel outside of a scan seems very secondary. I am not sure if I will implement it.